motioneye-project/motioneye #72: Motioneye service stays stopping and starting

enieuwenhuis, 2015-12-10:
Hi
I tried several times to install motioneye raspbian. 
I get no errors on service motioneye start
But, it does not work. The service keeps on stopping and starting. See this log, any idea whats wrong? 
Best regards
Erik 

Dec 10 22:40:40 raspberrypi systemd[1]: Unit motioneye.service entered failed state.
Dec 10 22:40:40 raspberrypi systemd[1]: motioneye.service holdoff time over, scheduling restart.
Dec 10 22:40:40 raspberrypi systemd[1]: Stopping motionEye Server...
Dec 10 22:40:40 raspberrypi systemd[1]: Starting motionEye Server...
Dec 10 22:40:40 raspberrypi systemd[1]: Started motionEye Server.
Dec 10 22:40:43 raspberrypi systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Dec 10 22:40:43 raspberrypi systemd[1]: Unit motioneye.service entered failed state.
Dec 10 22:40:44 raspberrypi systemd[1]: motioneye.service holdoff time over, scheduling restart.
Dec 10 22:40:44 raspberrypi systemd[1]: Stopping motionEye Server...
Dec 10 22:40:44 raspberrypi systemd[1]: Starting motionEye Server...
Dec 10 22:40:44 raspberrypi systemd[1]: Started motionEye Server.
Dec 10 22:40:47 raspberrypi systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Dec 10 22:40:47 raspberrypi systemd[1]: Unit motioneye.service entered failed state.
Dec 10 22:40:47 raspberrypi systemd[1]: motioneye.service holdoff time over, scheduling restart.
Dec 10 22:40:47 raspberrypi systemd[1]: Stopping motionEye Server...
Dec 10 22:40:47 raspberrypi systemd[1]: Starting motionEye Server...
Dec 10 22:40:47 raspberrypi systemd[1]: Started motionEye Server.

ccrisan, 2015-12-11:
Have you tried manually starting motionEye: meyectl startserver -d?

enieuwenhuis, 2015-12-11:
I have just entered the command sudo meyectl startserver -d
But that makes no difference. No error when I give meyectl startserver -d

ccrisan, 2015-12-11:
What's the output that you get? Does it work or not?

enieuwenhuis, 2015-12-11:
There is no output on the command meyectl startserver -d
The prompt returns, without output.

chrisf4lc0n, 2015-12-12:
Are you using the new pre-compiled by Calin ffmpeg?

enieuwenhuis, 2015-12-13:
Yes I do use the pre-compiled ffmpeg

ccrisan, 2015-12-13:
How about `motion -h`?

enieuwenhuis, 2015-12-13:
There is no output on that command
[image]

ccrisan, 2015-12-13:
Have you followed the procedure at step 1 from [these instructions](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)?

enieuwenhuis, 2015-12-15:
Yes, I followed those instructions at step 1, and other steps.

ccrisan, 2015-12-15:
Have you had the *deb-multimedia* repo installed on your system before? If yes, have you removed it and then reinstalled the packages `libavcodec-extra-56`, `libavformat56`, `libavresample2` and `libavutil54`?

enieuwenhuis, 2015-12-17:
Yes I had deb-multimedia repo installed. I removed this line from /etc/apt/sources.lst
Then I did sudo apt-get update
then I did sudo apt-get install --reinstall libavcodec-extra-56 libavformat56 libavresample2 libavutil54

the problem still exists. I don't know it anymore...

chrisf4lc0n, 2015-12-18:
Removing the repo from the repos list is not enough. You would need to:
sudo apt-get purge ffmpeg
sudo apt-get remove libavcodec-extra-56 libavformat56 libavresample2 libavutil54
sudo apt-get autoremove
Then you can start installing everything step by step.

ccrisan, 2015-12-18:
@enieuwenhuis Indeed, your command looks a bit different than the one mentioned in the wiki. Please use @chrisf4lc0n's instructions and see if it works.

enieuwenhuis, 2015-12-23:
Thanks for your support. I tried this but the result is the same. 
[image]

I don't know it anymore, the problem is not your support ! You can close this issue, if you want.

ccrisan, 2015-12-23:
The only suggestion I have left is to start over with a clean and up-to-date Raspbian.

enieuwenhuis, 2015-12-23:
I will try fresh installation during Christmas. Thanks and best regards.

chrisf4lc0n, 2015-12-23:
If you do not use your Raspberry Pi for anything else than MotionEye the MotionEye OS would be even better option. It pretty much works out of the box with most of the cameras...

enieuwenhuis, 2015-12-24:
Hello, a fresh installation of Raspbian is the solution for me. It works. Thanks.

ccrisan, 2015-12-24:
Great! I'll close the issue then.

closed 2015-12-24 by ccrisan

chrisf4lc0n, 2015-12-24:
There must have been some ffmpeg leftovers on your system preventing you from getting it work properly.