motioneye-project/motioneye #73: [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

jeanmarc77, 2015-12-12:
Hello, 

Sorry to bother you again and again.. I try a wifi camera and this one fill my log even though i set log_level to quiet. Do you know how to shut up that warning for good ? 
Thanks

journalctl -f
-- Logs begin at Sat 2015-12-05 14:05:45 CET. --
Dec 12 18:21:38 i3 motion[18546]: [1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Dec 12 18:21:38 i3 motion[18546]: [1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Dec 12 18:21:58 i3 motion[18546]: [1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Dec 12 18:21:59 i3 motion[18546]: [1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

ccrisan, 2015-12-12:
Unfortunately this is motion that complains and not motionEye. I have just commited a change that will make motion follow motionEye's log level as much as possible. Starting with 0.29.1 you should no longer see these lines in your log when using a log level of `error`, `fatal` or `quiet`.

closed 2015-12-12 by ccrisan

jeanmarc77, 2015-12-12:
Thanks indeed !