motioneye-project/motioneye #74: 'run a command' on motion detection

vdagamb, 2015-12-13:
Hello Crisan,
at first thanks for the great piece of software, just made a donation las night ;-)
I have been using motioneye with the raspberry (RPi2) and connected another raspberry and a webcam to watch the animals in the garden. certainly my wife is very keen to have a view at the pictures in the morning. I have been using it for 6 months now. I use a Raspberry Pi 2 with motioneye v. 0.24 - was too lazy so far for completely reinstall. 

Now my question: On another Raspberry Pi (RPi1) 
I want to switch a socke by a shell command defined in the moetioneye admin web interface.
The command is 
ssh pi@192.168.178.59 " /usr/local/bin/send 00001 3 1 &"
I had exchanged the certificates between  RPi1 and RPi2 i.e. the ssh command works at the command line without password input.   
Nevertheless, the same command, generated by motioneye at motion detection, gives and error message in motion.log:

Host key verification failed

Maybe motioneye is run by another user? 

It would be great to get this solved.

Have a nice Sunday,
Bernd

ccrisan, 2015-12-13:
Do you use motionEye with Raspbian? Or have you installed motionEyeOS? Or perhaps an older version of motionPie?

vdagamb, 2015-12-13:
Oh that is a quick reply :-)
yes, I work with raspibian 
motioneye 0.24 .
Raspbian GNU/Linux 7 
BR,
Bernd

ccrisan, 2015-12-13:
Well then this is definitely not something that motioneye controls in any way. motionEye runs with the user you started it (probably with `pi`). But whenever you reinstall a Linux OS, the ssh keys need to be regenerated (unless you back up the ~/.ssh folder).

vdagamb, 2015-12-13:
Thanks Crisan, for the hint, I wanted to copy it anyway to a flashdisk. Any hint where I could search? As user Pi direct input it works and when the same string is triggered through motioneye  - Host key verification failed ?? Any Idea whereelse to search?
Thanks
Bernd

ccrisan, 2015-12-13:
How do you start motionEye?

vdagamb, 2015-12-13:
In the mean time I found out, that the command rather shold be
ssh 192.168.178.59 " /usr/local/bin/send 00001 3 1 &"
because using ssh pi@... I had 2 users Pi 
Moreover, from the command line it works too excluding the &
and putting the quotes like this:
ssh 192.168.178.59  /usr/local/bin/send "00001 3 1"

Start motion eye: 
in /etc/rc.local there is one line
/home/pi/motioneye/motioneye.py & 

that starts the program,
ps shows
 399 ?        S      2:49 python /home/pi/motioneye/motioneye.py
 2556 ?        Sl    11:13 /usr/bin/motion -c /home/pi/motioneye/conf/motion.conf -n -d 1

I had another camera connected to an USB port which is not used in the moment.

/var/log/syslog:

Dec 13 16:26:12 raspberrypi2 rc.local[390]: 2015-12-13 16:26:12: ERROR: could not open camera config file: [Errno 2] No such file or directory: '/home/pi/motioneye/conf/thread-1.conf'
Dec 13 16:26:12 raspberrypi2 rsyslogd-2007: action 'action 17' suspended, next retry is Sun Dec 13 16:26:42 2015 [try http://www.rsyslog.com/e/2007 ]
Dec 13 16:26:12 raspberrypi2 rc.local[390]: 2015-12-13 16:26:12: WARNING: ignoring event for remote camera with id 1 (probably removed)
Dec 13 16:26:57 raspberrypi2 rc.local[390]: 2015-12-13 16:26:57: ERROR: could not open camera config file: [Errno 2] No such file or directory: '/home/pi/motioneye/conf/thread-1.conf'
Dec 13 16:26:57 raspberrypi2 rsyslogd-2007: action 'action 17' suspended, next retry is Sun Dec 13 16:27:27 2015 [try http://www.rsyslog.com/e/2007 ]
Dec 13 16:26:57 raspberrypi2 rc.local[390]: 2015-12-13 16:26:57: WARNING: ignoring event for remote camera with id 1 (probably removed)

vdagamb, 2015-12-13:
This is the motion.cfg 
(it does not point to thread-1.conf which actually does not exist in ~/motioneye/conf)

# @enabled on
# @show_advanced on
# @normal_password
# @admin_username admin
# @admin_password
# @normal_username user

control_port 7999

thread thread-2.conf

webcontrol_port 7999

vdagamb, 2015-12-13:
one more thing: 
sudo find / -name thread-1.conf 
does not show any result

ccrisan, 2015-12-13:
Well you don't have a thread-1.conf and that is not a problem. You probably added two cameras and then removed the first one.

vdagamb, 2015-12-13:
Yes, this might be the case. Actually I had 3: one in the motioneye machine, another in another raspi and a usb-webcam.

what about the start procedure? : 
Start motion eye: 
in /etc/rc.local there is one line
/home/pi/motioneye/motioneye.py &

that starts the program,
ps shows
399 ? S 2:49 python /home/pi/motioneye/motioneye.py
2556 ? Sl 11:13 /usr/bin/motion -c /home/pi/motioneye/conf/motion.conf -n -d 1

ccrisan, 2015-12-13:
If you start motionEye from /etc/rc.local (which is fine), it runs with `root` instead of `pi`. You should probably consider generating those ssh keys for the user `root` as well.

vdagamb, 2015-12-13:
NOW IT WORKS!!
That was the issue ... I have learnt a lot today :-)
have a nice evening,
Lots of thanks,
Bernd

ccrisan, 2015-12-13:
That's great to hear! You're welcomed.

closed 2015-12-13 by ccrisan

vdagamb, 2015-12-13:
just one off-topic question: I will have to up-date to ver 0.27 some time, to get the regular up-dates. ist it possible, to maintain the ver 0.24 for a while, say by renaiming the ver 0.24 to motioneye.old , then install the new version from scratch, copy the config-file and when it works fine delete the old version?

ccrisan, 2015-12-13:
Don't. The differences are huge. I strongly advise you to start clean.

frazt, 2016-01-14:
I have dlink IP cams connected and visible however motion detect does not seem to be functioning.
I have everything selected as far as the settings go and still no luck.
I have messed with ip cams quite a bit, along with Zoneminder.  I am hopeful for this to work to replace zoneminder. Any suggestions?

ccrisan, 2016-01-15:
Please attach your configuration files (removing sensitive data).

thom13, 2016-11-15:
Hi,
I have installed motioneye os on a RPI2, that's work well.

But I try to interact with a switch in domoticz with this command:

`![image](https://cloud.githubusercontent.com/assets/20850448/20321481/963a6098-ab43-11e6-9365-718ee1d1590f.png)`

This command work in the terminal of motioneye os, but doesn't work in /Motion Notifications/Run a command.

ccrisan, 2016-11-15:
Use a shell script for that, as the command contains a lot of special characters that would probably need escaping.

skcareer, 2017-05-10:
Hi Ccrisan,

I have installed motioneye on raspbian_os so that I can use some inherent capabilities of the os with motioneye. But nothing happened when I input the directory of my pyhton script under the Motion Notification  "run command". Could you please assist me with this?

Thanks in advance.

Regards,

skcareer, 2017-05-10:
Its my pleasure to receive help from anyone who has resolved issue like this before.
Thanks in anticipation.

ccrisan, 2017-05-10:
@skcareer you should fill in the full path to the python script, not just the directory! Also, make sure it's executable and that it works well outside of motionEye.

skcareer, 2017-05-10:
Hi Ccrisan,

Thanks for your promt response.

I have created another smaller program with twitter API(just to tweet
'Hello World" to my twitter account). This was tested to be working fine on
my raspberry pi 3.

I typed in the script path as follow on the Motion Notification *run
command* textbox:
* /home/pi/twitter1.py*
The python script file name is twitter1.py and its saved on pi directory
(folder).

Nothing happens when I simulated motion in front of the raspberry pi camera.

Is there anything I ought to have done that I have not done?

Thanks in advance.

Best Regards,
Sunkanmi.

LolloMazzy, 2017-12-06:
In which folder should I put the bash file for the commands to be used MOTION NOTIFICATION?

Use MotionEye on RaspianOS

kduytsch, 2018-07-13:
Got same problem, script are always "Not Found", while they work fine executed manually. I tried to put them at different places (even if I use full path), but no luck

 &: 1:  &: motionDectection.sh: not found

 &: 1:  &: /etc/motioneye/motionDectectionCamera1OFF.sh: not found

 &: 1:  &: /bin/bash /etc/motioneye/motionDectection.sh Camera1 OFF: not found

ccrisan, 2018-07-13:
@kduytsch what's `&: 1: &`?

kduytsch, 2018-07-13:
That's a copy/paste from the motion.log log file

[6407704:motion] [NTC] [ALL] motion_startup: Logging to syslog
[6407704:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [WRN] [NET] netcam_next: called with no data in buffer
 &: 1:  &: motionDectectionCamera1OFF.sh: not found
 &: 1:  &: motionDectectionCamera1OFF.sh: not found

ccrisan, 2018-07-13:
And are you 100% sure the script is present at `/etc/motioneye/motionDectectionCamera1OFF.sh` and is *executable*?

kduytsch, 2018-07-13:
root@raspberrypi:/etc/motioneye# ls -l motionDectectionCamera1OFF.sh
-rwxr-xr-x 1 root root 223 Jul 13 20:58 motionDectectionCamera1OFF.sh

But I just found the problem.... The ^M at the end of the file. In Unix/Linux, usually when you edit with vi, you see them. Was not the case here. I edited via a text editor and changed the EOL from dos to unix and now it does work. Still can't understand why it was working on CLI

kduytsch, 2018-07-13:
So now, I can run a script with full path and even CLI arguments like :
/etc/openhab2/scripts/motionDectection.sh Camera1 ON