motioneye-project/motioneye #75: Error with pip install motioneye

taeraeyttaejae, 2015-12-16:
i am using raspberry pi 2
i have followed your install instructions on https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

i am running into problems with :
pip install motioneye

the error msg on the end is:
    ValueError: --enable-jpeg requested but jpeg not found, aborting.

    ----------------------------------------
Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-Cgga48/pillow/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-j4514X-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Cgga48/pillow

----
outside of your instructions, i have downloaded pip with command -
sudo apt-get install python3-pip

ccrisan, 2015-12-16:
`python3-pip` shouldn't be used as it's part of python3. motionEye works with python2. Have you installed `libjpeg-dev`?

taeraeyttaejae, 2015-12-16:
i am beginner with this stuff, i followed your manual to the point but that wasn't mentioned there. i forgot to mention i am using debian-lite, if that matters.

taeraeyttaejae, 2015-12-16:
installed the python 2 with
sudo apt-get install python-pip

then tried to
sudo pip install motioneye

got error results down here

-----------
Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-act2iI/pillow/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-6T7Tvq-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-act2iI/pillow

chrisf4lc0n, 2015-12-16:
Try sudo apt-get install python-dev

ccrisan, 2015-12-16:
@taeraeyttaejae yes it is mentioned. Check out step number 3 of the [installation instructions article](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian). You have there `python-pip` as well as `python-dev` and `libjpeg-dev`.

taeraeyttaejae, 2015-12-16:
i think i had forgot sudo from the beginning of part 3 package install. something new was installed. i hope i get this working now :)

so, my mistake! the instructions are very easy to follow.

ccrisan, 2015-12-16:
No worries, I'm glad you got it working.

closed 2015-12-16 by ccrisan