motioneye-project/motioneye #76: Record from one camera when motion is detected on another camera

algestam, 2015-12-17:
I am using an IP camera that provides two rtsp streams, one in high resolution (1920x1080) and another in lower resolution (320x160). I would like to configure motioneye to only enable motion detection on the low-res stream (to save some CPU cycles) and when motion is detected, trigger recording or saving of still images from the high-res stream.

Is this possible to configure today or is it a new feature?

ccrisan, 2015-12-18:
I wouldn't know how to configure motion so that it uses two streams. This is rather a motion question than a motionEye one.

chrisf4lc0n, 2015-12-18:
What I would advise you to do is to use the low res stream with MotionEye so it records when movement appears on the stream. In the same time you can connect to high res stream through the VLC using rtsp protocol.

algestam, 2015-12-18:
Ok, thanks for the info :) I will look into motion to see if I can configure it to accomplish what i want to do.

closed 2015-12-26 by ccrisan