motioneye-project/motioneye #77: Camera Board

RoboGrinch, 2015-12-17:
Hi, could you give me any help on how to get the raspberry camera board to work with motioneye please? 

Really love MotionEye its a great bit of software, working perfect with a few USB cameras, would love to get it working with the camera module now.

Thanks.

sgiannulli, 2015-12-17:
not sure about what's your problem adding a CSI picamera on motioneye camera list...try:
sudo nano /etc/modules
add the following at the end of the file: bcm2835-v4l2 and reboot
as stated on the wiki about how to install on raspbian......of course if you're using raspbian
or manage it using raspi-config command https://www.raspberrypi.org/documentation/usage/camera/README.md

krovs, 2015-12-19:
I have the same problem, my camera is working, I added it to /etc/modules but in motioneye I get (no cameras)

chrisf4lc0n, 2015-12-20:
Mind that RasPi 1 is not that powerful, so I would normally advise to reduce the resolution to 640x480 and 20 FPS, anything more can cause crashing or at least stuttering it does with me...

flancrust, 2015-12-20:
out of curiosity what settings have you got for your pi cam? i have a pi2 and have max resolution 1600x1200 only set to 2fps. mainly as on my pib+ it struggled as mentioned with anything usable.

chrisf4lc0n, 2015-12-21:
@flancrust My example is a bit more complicated, as I am using h264_v4l2_rtspserver just to stream from my Pi to main server which is running Ubuntu 14.0.4.3 and all the MotionEye recording and motion detection happens there. The Rasberry Pi only streams the actual image from the camera, currently running it at 640x480 in 20 FPS and it seems to keep that up, I am sure it could stream more than that, but I am also running RFSniffer which seems to be taking most of the processing power. I am planning to either replace Pi 1 with 2 or use my Banana Pi Pro for the task.
I need such setup as I am also using the Pi as a Doorbell, which takes photos of people ringing it and sends them to me in the same time streaming the image to the main server...
[image]

flancrust, 2015-12-21:
@chrisf4lc0n nice set up, mines purely for home cctv so pi2 doing all the grunt work(camera load and server for remote access) as it is a low power solution that can be left on 24/7, looking into foscam for my garage or maybe my pi1 as stream provider but need to investigate a solid ir solution. foscam can be bought for £40 but figure a cheap ir webcam could achieve a good enough result for much cheaper.

If I used fast net cam on pi1 and stream it to pi2 could pi2 handle motion for the pi1 cam? therefore giving me a high frame rate at a decent resolution.

chrisf4lc0n, 2015-12-21:
@flancrust When doing some extensive testing with my Banana Pi Pro, which is just dual core vs quad Pi 2 I can confirm that the Banana can easily handle 2 cameras at 640x480 with 20 FPS, but the Banana Pi Pro has a big advantage over Pi the 1Gbit Ethernet, which is not bottlenecking anything.
For a little bit extra you can get ODROID-XU4 or CuBox-i 4×4 which are much more future proof, both sporting Gigabit Ethernet.
For the garage I would recommend Sinocam, depending on the resolution the price is between 20 to 30 odd quid. I currently have got 2, one VGA and one 1 1080p, both have going strong for nearly 2 years now.
I hope that helps.
Chris.

ccrisan, 2015-12-21:
For the record, I also have been using Odroid XU4 with motionEye (on Arch Linux) for a month or so, with 6 netcams (motionEyeOS on RPI B+, Fast Network Camera). I couldn't be happier about the experience.

chrisf4lc0n, 2015-12-21:
You cannot really beat octa core CPU and Gigabit Ethernet! The Odroid XU4 is by far the most powerful out of the micro single board computers! Anything more than 4 cameras would struggle on any other config...

flancrust, 2015-12-21:
I would love an odroid but can't justify the price for my use. at most I will have 3 cams so no need for all that muscle. plus the additional power use, whilst not a huge amount, for a surveillance set up running 24/7 is unnecessary. I have my rpi2 running motioneye via wheezy as i had issues getting wifi working on motionos. then I have a spare rpi B to play around with, 

ordered a cheapo chinese ip camera to have a play round with before I invest any really money. nothing beats china orders, waiting a time frame between 7 and 35 days when randomly when you have forgotten all about an order your package arrives.

By the way Cailin, top notch system!

closed 2015-12-26 by ccrisan