motioneye-project/motioneye #78: RTSP camera test does not handle authentication

ccrisan, 2015-12-20:


chrisf4lc0n, 2015-12-20:
I have got 2 RTSP cameras one with authentication and do not seem to have that problem...

ccrisan, 2015-12-20:
I believe that some RTSP cameras respond with 401 when queried for stream information if no authorization info is provided. Could you please temporarily enable public (but password-protected) access to one of your RTSP cameras so I can run some tests? If course you could email me the details in private.

chrisf4lc0n, 2015-12-20:
I can leave the stream up as long as you need it mate, just let me know when you are done. Also it might be cut off tomorrow for some time as I am changing my ISP... I will do my best to have it up and running ASAP, but cannot guarantee any time frames, as I will not be around for about half a day.

chrisf4lc0n, 2015-12-21:
As mentioned earlier I am with no Internet, once connected to the new ISP I will give you the access to my RTSP camera again.

ccrisan, 2015-12-21:
Sure, no worries.

chrisf4lc0n, 2015-12-25:
Sorry I could not help more...

ccrisan, 2015-12-26:
I tried to set up a ffserver-based RTSP stream that streams captured video from a v4l2 device. Everything seems to work just fine apart from the fact that I couldn't figure out how to configure authentication for the stream. Does RTSP support authentication anyway?

ektus, 2015-12-26:
I've got a network camera that requires authentication to view the RTSP stream. VLC asks for credentials, too, as soon as I try to open the stream. I was able to use the camera with Motioneyeos and have switched to Motioneye today, with no problems at all as soon as streaming address, username and password were entered correctly. Using current 0.30rc2 on a RPi2 with Raspian Wheezy. I'm on a limited volume LTE (4G mobile) connection here, so can't provide a stream externally.

chrisf4lc0n, 2015-12-26:
@ektus even if you are on the limited bandwidth Calin should not need too much of it to get the things working... He only needs the authentication moment, not the whole stream..., which should not take more than couple of Megs of your data...

ektus, 2015-12-26:
I'm not so sure it's even possible to get incoming connections working here, but I could provide a wireshark log of the connection between vlc and the camera. Contact me by e-mail if you like.
Well, and for me authentication is working, or would this be needed earlier in the camera setup process?

The portion with the 401 error reads as follows: 
```
0030   0e 42 36 96 00 00 52 54 53 50 2f 31 2e 30 20 34  .B6...RTSP/1.0 4
0040   30 31 20 55 6e 61 75 74 68 6f 72 69 7a 65 64 0d  01 Unauthorized.
0050   0a 43 53 65 71 3a 20 33 0d 0a 44 61 74 65 3a 20  .CSeq: 3..Date: 
0060   53 61 74 2c 20 44 65 63 20 32 36 20 32 30 31 35  Sat, Dec 26 2015
0070   20 32 30 3a 33 33 3a 30 31 20 47 4d 54 0d 0a 57   20:33:01 GMT..W
0080   57 57 2d 41 75 74 68 65 6e 74 69 63 61 74 65 3a  WW-Authenticate:
0090   20 44 69 67 65 73 74 20 72 65 61 6c 6d 3d 22 4c   Digest realm="L
00a0   49 56 45 35 35 35 20 53 74 72 65 61 6d 69 6e 67  IVE555 Streaming
```

Ded1er, 2016-01-13:
With Mr Dave's motion it's working

ccrisan, 2016-01-13:
@Ded1er do you have a RTSP camera that requires authentication?

Ded1er, 2016-01-14:
@ccrisan Yes, I have a Foscam FI9803EP that requires authentication for the use of RTSP. 

At first I installed motion using the official Debian Jessie repository but it didn't work. By using MR Dave's motion, everything works. Thank you for Motioneye, it's really helpful!

ccrisan, 2016-01-15:
@Ded1er would you by any chance be willing to provide me with temporary remote access to your password-protected RTSP stream (you could email me the details)?

Ded1er, 2016-01-20:
I'm sorry but the camera is for a professionnal use at my company. I am not allowed to give you even a temporary access.

ccrisan, 2016-01-20:
No worries. But just out of curiosity, could you please try to remove and re-add your RTSP camera without supplying any username and password, and then let me know what happens?

Ded1er, 2016-01-20:
There are no errors while adding the camera without supplying login/password but I get this grey screen. 

[image]

ccrisan, 2016-01-20:
Yes, that's what I was afraid of. In theory I could detect an authentication problem from the moment details are present in the Add Camera Dialog. The RTSP protocol appears to be very similar to HTTP so the standard HTTP authentication headers could be used. Unfortunately in practice they don't seem to be used so performing additional checks at that point is useless.

Thanks anyway for your time. I'll mark the issue as *won't fix* now.

closed 2016-01-20 by ccrisan

bwong, 2016-04-17:
I know that this issue has been filed under wontfix, but does anyone within the thread have any experience with Amcrest Cameras? I am seeing the `rtsp netcam returned erroneous response: RTSP/1.0 401` message when trying to add a camera. I was able to view the stream using VLC, using both a user that needs authentication and one that does not require authentication (blank password).  @ccrisan Does this basically have to do with your belief that some RTSP cameras respond with 401 when queried for stream information if no authorization info is provided?

ektus, 2016-04-17:
I've got a QCAM PoE bullet camera, which is an Amcrest brand, and I've got it working with motioneye. Well, the camera resolution doesn't match any of the available resolutions within motioneye, and there are lots of garbled images, but other than that, it's basically working. I had to enter the password and then the username, so the login credentials were complete upon connecting to the camera, but that's it. Haven't touched it for months now.

chrisf4lc0n, 2016-04-17:
@ektus
You can edit one of the files to change the resolutions in MotionEye...
ccrisan could you point the man into the right direction as I cannot remember which file had the res in it and I am at work so cannot check it.

ektus, 2016-04-17:
I've done so already, but that modification gets overwritten as soon as you make any changes to the camera from the GUI. It would be necessary to alter the preset values, but as it's working for me just good enough, I won't bother further.

bwong, 2016-04-17:
@ektus Thanks for the info.  I have an Amcrest IPM-721, I wonder if it would be different per build/firmware updates from Amcrest.  Did you have to do anything different? Or did it just work out of the box?

ektus, 2016-04-17:
It did work out of the box. There are no firmware updates available anyway, at least not that I know of.

bwong, 2016-04-18:
What was the URL format you used for your camera?

I am using this for my Amcrest

`rtsp://<ip-address>:554/cam/realmonitor?channel=1&subtype=1`

ektus, 2016-04-18:
More something like rtsp://<ip-address>:554/h2_64Preview_01_sub but I'll have to check the exact settings next weekend.

ektus, 2016-04-23:
The entire URL reads
`rtsp://192.168.0.12:554/h264Preview_01_sub`
This is for the low-res (640x360) preview stream I'm using for motion detection in order to greatly reduce load on the RPi2. I've got a Synology NAS running in parallel that records the main stream 24/7.

Regards
Ektus.

bwong, 2016-05-01:
Thanks @ektus.  Sorry for the late response.  If anyone is interested, I was able to find this documentation from Amcrest website.  I am still unable to use RTSP, but using the doc I was able to find and use the HTTP url to finally add my network camera.

Here's the documentation link:  https://s3.amazonaws.com/amcrest-files/AMCREST_CGI_SDK_API.pdf