motioneye-project/motioneye #79: SVG images have wrong content-type

jypma, 2015-12-21:
I've installed Motioneye 0.29.1 on arch linux through [AUR](https://aur.archlinux.org/packages/motioneye/) (which downloads the release). Running on python2-tornado 4.3.0-1.

The UI and backend works, but any SVG images are served up with `Content-Type: application/octet-stream`, causing Firefox to not show them:

```
curl -v http://localhost:8765/static/img/motioneye-logo.svg
*   Trying ::1...
* connect to ::1 port 8765 failed: Connection refused
*   Trying 127.0.0.1...
* Connected to localhost (127.0.0.1) port 8765 (#0)
> GET /static/img/motioneye-logo.svg HTTP/1.1
> Host: localhost:8765
> User-Agent: curl/7.46.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Length: 1813
< Accept-Ranges: bytes
< Server: TornadoServer/4.3
< Last-Modified: Sun, 30 Aug 2015 12:39:33 GMT
< Etag: "8caecd52deecadcf696adf3dc1e09816"
< Date: Mon, 21 Dec 2015 08:09:15 GMT
< Content-Type: application/octet-stream
< 
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
...data...
```

It's probably tornado's web server having a messed up MIME type registry somewhere; does anyone know where to diagnose this?

jypma, 2015-12-21:
Well, that was quick. Python's `mimetypes` module depends on (amongst other possibilities) apache being installed so that `/etc/httpd/conf/mime.types` exists. Why on earth python wouldn't bundle their own is a mystery.

Anyways, not a Motioneye issue obviously.

closed 2015-12-21 by jypma

melchips, 2015-12-31:
I had the same issue on archlinux but installing extra/mime-types package and restarting motioneye have fixed it.