motioneye-project/motioneye #80: Displaying camera feeds on small screens

chrisf4lc0n, 2015-12-21:
Hi,
I have noticed that you have added some more options for camera feed layouts, unfortunately it does not seem to be doing the job on my small Xubuntu Lenovo X61s screen. Everything looks rather weird and out of place, the old options from 0.30rc1 seemed to be much better for my little screen.

chrisf4lc0n, 2015-12-22:
Just went back to:
sudo pip install git+git://github.com/ccrisan/motioneye.git@5b78d91c2040d7383b30a0f0d731ff6809d2068e
commit.

ccrisan, 2015-12-22:
If you're talking about the full screen layout in 0.30rc2 vs 0.30rc1, I know about that regression - I experienced it myself :)

chrisf4lc0n, 2015-12-22:
That I exactly what I meant... When I switch the tool view on the side of the camera feeds the feeds themselves do not seem to scale, they windows just get smaller with a small piece of viewable area from the feed.

ccrisan, 2015-12-22:
It'll be taken care of in time for 0.30 final release.

closed 2015-12-25 by ccrisan