motioneye-project/motioneye #81: Protocol rtsp not supported

thinkdao, 2015-12-23:
INFO: hello! this is motionEye server 0.30rc2
[image]

TheHolm, 2015-12-23:
What "motion" are you using? It seems you are using "official" one. Id does not support rtsp. Grab one from git.

thinkdao, 2015-12-23:
@TheHolm , i'm using https://github.com/Mr-Dave/motion and standalone motion working with rtsp.

chrisf4lc0n, 2015-12-23:
@thinkdao What OS/hardware?

thinkdao, 2015-12-23:
@chrisf4lc0n, Gentoo/Xiaomi XiaoYi,  but all rtsp links doesn't work

ccrisan, 2015-12-23:
@thinkdao what does "motion -h" say?

chrisf4lc0n, 2015-12-23:
@thinkdao Did you build it yourself? I assume there were no errors? Give the SVN version a go and report back if you can.

thinkdao, 2015-12-23:
@chrisf4lc0n , hm, when building from setup.py - doesn't work rtsp and no show errors. When install it from pip all okay.

chrisf4lc0n, 2015-12-23:
Sorry, I was not specific enough... Did you build the motion from the source?

thinkdao, 2015-12-23:
@chrisf4lc0n , yeah, all packages in gentoo building from sources.

chrisf4lc0n, 2015-12-23:
Yup SVN supports RTSP, also what is the output of Calin's question?

chrisf4lc0n, 2015-12-23:
Also once you have built it issue ./motion -h from the folder you built it...

thinkdao, 2015-12-23:
@chrisf4lc0n 
>$ motion -h
>motion Version Unofficial-Git-2caced3, Copyright 2000-2005 Jeroen Vreeken/Folkert van >Heusden/Kenneth Lavrsen
>
>usage:  motion [options]
>
>
>Possible options:
>
>-b                      Run in background (daemon) mode.
>-n                      Run in non-daemon mode.
>-s                      Run in setup mode.
>-c config               Full path and filename of config file.
>-d level                Log level (1-9) (EMG, ALR, CRT, ERR, WRN, NTC, INF, DBG, ALL). default: 6 / NTC.
>-k type                 Type of log (COR, STR, ENC, NET, DBL, EVT, TRK, VID, ALL). default: ALL.
>-p process_id_file      Full path and filename of process id file (pid file).
>-l log file             Full path and filename of log file.
>-m                      Disable motion detection at startup.
>-h                      Show this screen.
>
>Motion is configured using a config file only. If none is supplied,
>it will read motion.conf from current directory, ~/.motion or /etc/motion.

thinkdao, 2015-12-23:
>Yup SVN supports RTSP, also what is the output of Calin's question?

yeah, svn version build without errors and rtsp working. 
> motion -h
>motion Version trunkREV561, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen

chrisf4lc0n, 2015-12-23:
Good to hear we got to the bottom of it... Initially I seemed to have a similar problem on the Ubuntu Server 14.04.3 with the Mr Dave version of motion but SVN worked with no issues, right now both of them work with no issues.
Regards Chris.

ccrisan, 2015-12-23:
Well is the issue really solved @thinkdao?

thinkdao, 2015-12-23:
@ccrisan , yeah, install via setup.py working only with SVN build motion for rtsp support. Add notice on wiki :+1:

closed 2015-12-23 by thinkdao