motioneye-project/motioneye #82: Reconnect for RTSP

thinkdao, 2015-12-23:
Doesn't reconnect and doesn't show gray background for offline cam. Picture only freeze.

chrisf4lc0n, 2015-12-24:
Hold on I am getting lost here... It does connect to the RTSP camera now, but it when connected no picture gets streamed?
Can you change the log level to debug in /etc/motioneye/motioneye.conf and post the output of the /var/log/motioneye.log covering all the sensitive data?

thinkdao, 2015-12-24:
@chrisf4lc0n yep, it connect to RTSP camera, but when has issue with network or camera going reboot - motioneye continues streaming video with frozen picture. 

debug lvl:

    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: creating mjpg client for camera 1
    DEBUG: mjpg client for camera 1 connected on port 8081
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: mjpg client for camera 1 on port 8081 has been idle for 10 seconds, removing it
    DEBUG: connection closed for mjpg client for camera 1 on port 8081
    DEBUG: mjpg client for camera 1 on port 8081 removed
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...
    DEBUG: running garbage collector for mjpg clients...

drewzh, 2016-03-16:
I think you might find that switching to motion-mrdave binary will work fine, I had the same issues when using the latest available motion build but have since switched and have no issues since. :+1:

ccrisan, 2016-03-17:
In some cases (usually RTSP-related) motion-mrdave binary works better. That's why motionEyeOS defaults to it.

closed 2016-03-17 by ccrisan