motioneye-project/motioneye #83: Add date to filename when saving to Google Drive

chaseacton, 2015-12-23:
Still images and videos saved to the local disk are configured to have the date and time as their filename. However, when saving a file using the Google Drive setting, filenames are formatted with just the time. (9:45 would be 09-45-00.jpg). This causes older images from previous days to be overwritten if they occurred at the same time.

ccrisan, 2015-12-24:
Yes, that is something that's not quite cool and needs to be revised.

stuartbloom, 2016-02-12:
+1 for this, infact, would it be possible to use the same format as that used by the file names?  Could folders also be created?

Thanks and keep up the GREAT work

chaseacton, 2016-02-12:
+1 for folders for each day. File management on Google Drive is a pain, so it would be really helpful if I could delete folders than have to delete each image.

For extra credit, it would be nice if within each day's folder, there are subdirectories for each camera based on their name. Basically, if someone breaks in, I want to be able to drill down to the day and then a specific camera. Looking though hundreds of images/videos based on their timestamp is time-consuming.

bovine, 2016-04-20:
As a workaround, you can change the settings under "Movies -> Movie File Name" to be "%Y-%m-%d-%H-%M-%S" and the files uploaded to Google Drive will now have the date in them.

It would still be nice to have separate date folders created to simplify bulk deletion though.

azadaffif, 2016-06-25:
Is it possible to automatically create hierarchy(folder) just like how it is stored in local repository (%Y-%m-%d/%H-%M-%S ) ? Thanks for the wonderful motioneye :)

ccrisan, 2016-07-03:
This should be fixed with 741fa88fc20d3dfbd55dd3fc4fdcb04acf8b8c8e

closed 2016-07-03 by ccrisan