motioneye-project/motioneye #84: Missing V4L2 controls

mvanbreeden, 2015-12-30:
Hi,

I found that two controls for my Raspi Cam are missing in the controls of _list_ctrls: iso_sensitivity and auto_exposure_bias.

I also found the problem: the re.match statement in line 408 of v4l2ctl.py. It only looks for lines where there is a separator before the ':'. With both mentioned controls, this is not the case (check with "v4l2-ctl -l").

The solution I used to circumvent is to add another match statement to find both types of controls (with and without separator before ':'

match = re.match('^\s*(\w+)\s+\(\w+\)\s+\:\s*(.+)', line)
        if not match:
            match = re.match('^\s*(\w+)\s+\(\w+\)+\:\s*(.+)', line) #extra check for controls with no separator
            if not match:
                continue

Do let me know what you think. 

And thanks for the great work you did! I've learned a lot going through the code. One thing I would like added (maybe will do so myself) is automatically switching enabled camera to night mode, for instance based on  "Calulating sunrise and sunset in Python" by Michel Anders

Best regards
Marcel

ccrisan, 2016-01-09:
Thanks for your feedback and for your contribution.

I have however a question regarding the extra v4l2 controls: supposing that `iso_sensitivity` and `auto_exposure_bias` would be "seen" by the regex, how would that help you? I mean the UI presents a limited set of controls which are hardcoded. So you wouldn't be able to tweak the iso_sensitivity anyway, would you?

ccrisan, 2016-01-09:
I have loosen up the regex a bit so that it also takes into account the no-spaces-before-a-colon case.

mvanbreeden, 2016-01-09:
Your fix is even better.
With regards to your question: I was looking into a way to implement night view. Thought that was doable by night scene or combination of iso sensitivity and other settings. Since I have since learned that motion is 'blocking' the. Other controls, it will not help directly. But then again: if it's a bug, we need to squash it ;-)

closed 2016-01-09 by mvanbreeden