motioneye-project/motioneye #85: V4L2.py: _set_ctrl messes up menu or intmenu values

mvanbreeden, 2015-12-30:
Hi,

in V4L2.py line 313, the statement:
      value = int(round(min_value + value * (max_value - min_value) / 100.0))
is fine for slider values (relative values), but messes up possible menu or intone values (which are exact).

My workaround was to add a relative=True argument to _set_ctrl, so that set functions that require that (set_iso will be a possibility) can call:
_set_ctrl(device, 'iso_sensitivity', 4, False) 

And the previous mentioned statement becomes:
       if relative:        
            value = int(round(min_value + value * (max_value - min_value) / 100.0))

Best regards
Marcel

Castor123, 2016-01-08:
Hi Marcel,

did you have a Workaround how i can add e.g. iso_sensitivity on motoneye.

Thanks.
Daniel

ccrisan, 2016-01-09:
This is in fact part of a bigger tasks, which would enable any v4l2 controls to show up in the UI. Some of them being of type "menu", they would be treated differently. Unfortunately the whole v4l2ctl module would need to be rewritten, as well as parts of the UI.

mvanbreeden, 2016-01-09:
I see your point. If you focus (for now) on just a few controls, maybe make that explicit that in this function (either in comment or explicitly coding). 

The thing is if people want to re-use or extend your code, they will run into this issue. Not important for motineye v0.30, but still ;-)

mvanbreeden, 2016-01-09:
Hi Castor,

I don't have one yet, since motion controls most settings. I saw some possibilities in other projects, but they do not use motion. I've been busy with other stuff, but will look into this for sure. I think it's a nice added feature that motineye can look into the dark as well.

Castor123, 2016-01-09:
Ok, thanks.

ccrisan, 2016-01-09:
@mvanbreeden I believe (although have not tested) that motion leaves unknown controls alone. That is, controls like iso_sensitivity or basically anything that's not available through the UI could be controlled externally via the v4l2-ctl command, while motion is running, or at startup.

mvanbreeden, 2016-01-09:
@ccrisan Thanks for the tip, I'll test and see. Would be the easiest way. If all else fails, a fork of motion might be necessary. One extra control to rule the night...

mvanbreeden, 2016-01-13:
I'm afraid the external route (using v4l2-ctl commands does not work. Even when restarting motion. So somehow v4l2 controls are set to defaults. I'll dive in the motion code, maybe I can find a better way.

closed 2016-01-13 by mvanbreeden