motioneye-project/motioneye #86: Question: why do files on my NAS have creation date 9 hours earlier?

mvanbreeden, 2015-12-30:
My NAS has proper time settings, the Raspberry does so also. Yet movies created have a creation date 9 hours earlier (I'm in Amsterdam timezone). I do not seem to find why...

ccrisan, 2015-12-31:
How about the timestamp that's overlaid on the video preview? Is it wrong as well? What have you set for Image File Name and Movie File Name.

mvanbreeden, 2015-12-31:
The overlaid timestamp is correct, and so are file creation timings for thumbs. I did not change defaults for filenames, which show correct time as well. It's just file creation data for avi's on NAS that's wrong

ccrisan, 2015-12-31:
Hmm well no idea about that. I never cared for the ctime/mtime attributes of files as they tend to be misleading. Are you running Raspbian? Have you rebooted since you changed your timezone?

mvanbreeden, 2015-12-31:
Yes I have rebooted, and running Raspbian. It just struck me that's all,
when looking in Finder. I'll check in the source myself, will keep you
posted. No biggie.

Being new to github, I wasn't able to label the issue as question... Sorry
for that

ccrisan, 2015-12-31:
Yes, please do :)

mvanbreeden, 2016-01-02:
I found out what the problem is,  will think about solution.
The problem is that the subprocess.open call for ffmpeg (make_movie) does not contain environment set. It assumes that environment variables will be gotten from parent process, but somehow those (or at least TZ) is not there.
The call to ffmpeg in make_movie_preview (check_out instead of Popen) does give environment info.

So probably they are called from different parent processes. Do you have any clue already? If not, I'll dive a little deeper ;-)

mvanbreeden, 2016-01-17:
I was ging about this wrong earlier. The problem with timezone is with starting motion with subprocess.popen, which apparantly does not give motion the right environment. Therefore motion will save the files without timezone info.

The best way to solve would be to exec as shell, what do you think @ccrisan?

ccrisan, 2016-01-18:
It's a valid question that you're rising here. I'll have to look into it and see how exactly the timezone is passed onto subprocesses. As far as I know, glibc will look for the /etc/zoneinfo file to deduce that information; the environment shouldn't have anything to do with it. But then again, I'm not sure about it and I'll have to run some tests.