motioneye-project/motioneye #88: Question: ffmpeg precompiled

sgiannulli, 2016-01-01:
https://github.com/ccrisan/motioneye/wiki/precompiled/ffmpeg_2.8.3.git325b593-1_armhf.deb
Hi,
just need to know if this has been compiled under raspbian jessie: I'm moving my RPi2 from wheezy to jessie and no more way to compile MrDave motion on a very fresh install
Thanks

ccrisan, 2016-01-01:
Yes, it's been compiled on a jessie setup.

sgiannulli, 2016-01-01:
[errori.txt](https://github.com/ccrisan/motioneye/files/75969/errori.txt)
never got those errors compiling motion on raspbian wheezy
could it be due to missing library dependencies?
Thanks

ccrisan, 2016-01-01:
Have you started with a clean installation? If not, please try to remove any previous ffmpeg-related packages, as indicated [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian#instructions). Then, you could try ./configuring and compiling motion again.

If that fails, either try to use this [prebuilt motion binary](https://github.com/ccrisan/motioneye/wiki/precompiled/motion-mrdave-raspbian) or try with a new clean and up-to-date Raspbian installation.

sgiannulli, 2016-01-02:
yes, every try has been made on a clean Rasbian Jessie install and every time the same error; according to me there's something conflicting between MrDave motion and your ffmpeg precompiled.
The solution was to use both the precompiled motion binary and ffmpeg deb packet you gave.
By now the only thing that sounds strange is that motioneye.log is not created while motion.log is and the only way to check if something is going wrong is having a look at syslog

closed 2016-07-03 by ccrisan