motioneye-project/motioneye #89: .AppleDouble directory errors

mvanbreeden, 2016-01-02:
Accessing files from a Mac on a network drive creates .AppleDouble directories. Which are hidden normally. However MotionEye does pick up these directories and also tries to scan them and add thumbnails. Since filenames in the .AppleDouble directories appear to be pictures and movies, the thumbnail handler will return errors and do unnecessary work.

Workaround/solution is to exclude all hidden, or more specific, .AppleDouble dirs from search paths (line 83 in media files.py)

ccrisan, 2016-01-03:
You're right, hidden folders should simply be ignored.

mvanbreeden, 2016-01-03:
I put in a pull request to suggest solution. Did focus on appledouble though, since that is automatically created without (most) users knowledge.

closed 2016-01-09 by mvanbreeden