motioneye-project/motioneye #93: Motioneye.service loaded failed failed

aleixpellicer, 2016-01-05:
I installed it following the Raspbian guide on the Wiki, on a freshly installed RASPBIAN JESSIE LITE.

This is the guide: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

On the guide I installed motion using the apt-get command

There were no errors during the installation but I can't access http://[ip of my raspberry]:8765 as it returns a connection refused.

When I do `systemcpl` it says the motioneye.service is load, active: failed, sub: failed

How can I check what is going on?

ektus, 2016-01-05:
I'm on Wheezy, but also had problems with systemcpl not working. I've got the motion autostart working with the following command in /etc/rc.local:
sudo meyectl startserver
For testing, the same command may be issued from a console (SSH or USB keyboard attached to the RPi).

aleixpellicer, 2016-01-05:
@ektus I did that, it outputs this:

```Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2876, in <module>
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 449, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 745, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 639, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: MarkupSafe```

ccrisan, 2016-01-06:
@aleixpellicer that's strange, as motionEye depends on Jinja2 which, in turn, depends on MarkupSafe. Therefore, when you have (hopefully) did a `pip install motioneye`, MarkupSafe should have been pulled as a dependency.

What you could do now is to try to install it manually: `pip install markupsafe`.

aleixpellicer, 2016-01-13:
I did a fresh install on Raspbian Wheezy and it was fine. Weird.

closed 2016-01-13 by aleixpellicer

fromdehart, 2017-08-12:
Just wanted to give you a shout out and say thanks! :)

I was having the same issue and installing MarkupSafe fixed it.