motioneye-project/motioneye #94: Feature request: ability to take snapshots from webui

jamox, 2016-01-06:
I regularly run into situations where I would like to get a snapshot of the camera image, now I just kill motion/motioneye and take the snapshot with raspistill...

There was some discussion about it in #41 but I didn't find an issue for the feature request.

Also it would be nice it the snapshot feature would be accessible via http/rest api.

ccrisan, 2016-01-06:
Yes, such a feature is planned, stay tuned!

Fudge0952, 2016-05-26:
Hey, just noticed this issue hasn't been touched in close to half a year. Any update on this feature request?

gsa390, 2016-07-22:
Is there a way to save snapshot jpg from Snapshot URL ? thanks

Samfox2, 2016-11-28:
curl -m 10 -o /tmp/snapshot.jpg http://192.168.x.xx:8765/picture/x/current

chrisf4lc0n, 2016-12-22:
@Samfox2 You have to have streaming enabled for that to work and the the URL has to be replaced with whatever is in the "Snapshot URL".

Cooper-Dale, 2017-02-03:
I have streaming enabled, but i got no image:

`pi@raspberrypi:~ $ curl -m 10 -v -o /tmp/aaa.jpg http://x:8765/picture/1/current/
* Hostname was NOT found in DNS cache
*   Trying 192.168.1.7...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Connected to 192.168.1.7 (192.168.1.7) port 8765 (#0)
> GET /picture/1/current/ HTTP/1.1
> User-Agent: curl/7.38.0
> Host: 192.168.1.7:8765
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 0
< Set-Cookie: motion_detected_1=false; Path=/
< Set-Cookie: monitor_info_1=33%26deg%3B; Path=/
< Set-Cookie: capture_fps_1=0.0; Path=/
* Server motionEye/0.35.1 is not blacklisted
< Server: motionEye/0.35.1
< Etag: "da39a3ee5e6b4b0d3255bfef95601890afd80709"
< Date: Fri, 03 Feb 2017 20:49:21 GMT
< Content-Type: image/jpeg
<
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
* Connection #0 to host 192.168.1.7 left intact
`

MrMontesa, 2017-05-04:
Hey all,
I came accross the same thing as I want to curl a "live" snapshot by URL. I got the same auth error as @Cooper-Dale and tried curl's auth implementation with _"curl -u user:pass"_, which doesnt work either.

When reading thought the issues here I came to know that Motioneye did not implement basic auth but rather a security token based auth. _curl -H "Authorization: Token xxxxxxxxxxxxxx"_ might help her but not too sure. I would need some help on how the string needs to look like.

Else official "curl" support to download a snapshot would be appreciated.
Any questions, let me know. Thanks

ccrisan, 2017-05-04:
One could use the snapshot url that's available from the settings panel, in the video streaming section. It includes the required signature.

MrMontesa, 2017-05-07:
Thanks for info.
I might be missing something then. My "Video streaming" section is completly empty.

[image]

I'm running a Pi Zero W with noir 2 cam. Motion is installed from the default repos.

ccrisan, 2017-05-07:
Just enable Advanced Options.

MrMontesa, 2017-05-07:
Thanks for your swift reply. From a comment in [#434](https://github.com/ccrisan/motioneye/issues/434#issuecomment-282489166 )  I assume the advanced options are only available in MotioneyeOS, right? I'm running the default raspbian.
Here's a screen from my settings pane.

[image]

chrisf4lc0n, 2017-05-07:
You have to flick the "Video Streaming" switch on and then you will see it.

ccrisan, 2017-05-07:
It's Advanced Settings in fact (mea cupla) and it's right there in your screenshot.

zZzimi, 2017-05-14:
Hi ccrisan!

Is there any solution to use different resolution at streaming and the snapshot? I want to use lower resolution if I do a stream and higher when I create a snapshot. Thanks your answer!

ccrisan, 2017-05-14:
@zZzimi unfortunately there's no such possibility. Currently motion only provides one stream.

vganyn, 2017-07-13:
@MrMontesa Can you provide working example please

MrMontesa, 2017-07-14:
@vganyn
1st, you have to get your token. The explanation how to get it is above.
2nd you can create a URL like this:
http://$YOUR_IP:8765/picture/1/current/?_username=admin&_signature=$your_token_goes_here
3rd: process the URL with whatever tool you need (e.g. curl, etc)

vganyn, 2017-07-14:
@MrMontesa I did 1st & 2nd. Problem with 3rd.(
```
curl -H "Authorization: Token $token" -v -o /tmp/aaa.jpg http://192.168.1.13:8765/picture/1/current/?_username=$user&_signature=$token
[1] 27563
$ *   Trying 192.168.1.13...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Connected to 192.168.1.13 (192.168.1.13) port 8765 (#0)
> GET /picture/1/current/?_username=$user HTTP/1.1
> Host: 192.168.1.13:8765
> User-Agent: curl/7.47.0
> Accept: */*
> Authorization: Token $token_here
> 
< HTTP/1.1 403 Forbidden
< Date: Fri, 14 Jul 2017 08:35:32 GMT
< Content-Length: 42
< Content-Type: application/json
< Server: motionEye/0.36
< 
{ [42 bytes data]
100    42  100    42    0     0   7193      0 --:--:-- --:--:-- --:--:--  8400
* Connection #0 to host 192.168.1.13 left intact
```

MrMontesa, 2017-07-14:
This is not HTTP auth, so you can ignore the -H parameter of curl. Secondly, iirc the user is always admin.
This command works for me:
`curl -o ./image.jpg 'http://192.168.23.17:8765/picture/1/current/?_username=admin&_signature=my_token'`

vganyn, 2017-07-14:
@MrMontesa Work like a charm. Thank you very much!

MrMontesa, 2017-07-21:
I have the problem that quite often curl exits with code 0 but did not download any image. In verbose mode it looks like this:
```
swa@nuc:~$ curl -vvv -m 10 -o /tmp/image.jpg 'http://192.168.23.17:8765/picture/1/current/?_username=admin&_signature=$my_token'
* Hostname was NOT found in DNS cache
*   Trying 192.168.23.17...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Connected to 192.168.23.17 (192.168.23.17) port 8765 (#0)
> GET /picture/1/current/?_username=admin&_signature=$my_token HTTP/1.1
> User-Agent: curl/7.38.0
> Host: 192.168.23.17:8765
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 0
< Set-Cookie: motion_detected_1=false; Path=/
< Set-Cookie: monitor_info_1=; Path=/
< Set-Cookie: capture_fps_1=0.0; Path=/
* Server motionEye/0.35.2 is not blacklisted
< Server: motionEye/0.35.2
< Etag: "da39a3ee5e6b4b0d3255bfef95601890afd80709"
< Date: Fri, 21 Jul 2017 08:06:54 GMT
< Content-Type: image/jpeg
<
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
* Connection #0 to host 192.168.23.17 left intact
swa@nuc:~$ ls -al /tmp/image.jpg
ls: Zugriff auf /tmp/image.jpg nicht möglich: Datei oder Verzeichnis nicht gefunden
```
As you can see the Download counters are all 0. When trying multiple times in a row with the same command, curl is suddenly able to get the image. This looks like this:

```
swa@nuc:/var/lib/openhab2$ /usr/bin/curl -vvv -m 10 -o /tmp/image.jpg 'http://192.168.23.17:8765/picture/1/current/?_username=admin&_signature=$my_token'
* Hostname was NOT found in DNS cache
*   Trying 192.168.23.17...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Connected to 192.168.23.17 (192.168.23.17) port 8765 (#0)
> GET /picture/1/current/?_username=admin&_signature=$my_token HTTP/1.1
> User-Agent: curl/7.38.0
> Host: 192.168.23.17:8765
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Length: 28635
< Set-Cookie: motion_detected_1=false; Path=/
< Set-Cookie: monitor_info_1=; Path=/
< Set-Cookie: capture_fps_1=2.1; Path=/
* Server motionEye/0.35.2 is not blacklisted
< Server: motionEye/0.35.2
< Etag: "bbdb5e21d61aedce55ed6dcab1e6375f03b449b7"
< Date: Fri, 21 Jul 2017 08:07:23 GMT
< Content-Type: image/jpeg
<
{ [data not shown]
100 28635  100 28635    0     0   565k      0 --:--:-- --:--:-- --:--:--  582k
* Connection #0 to host 192.168.23.17 left intact
swa@nuc:~$ ls -al /tmp/image.jpg
-rw-r--r-- 1 swa swa 28635 Jul 21 10:08 /tmp/image.jpg
```
The Download counters are properly filled with data and the image got downloaded.

Is it possible to increase logging on motioneye end to see whats going on?
Thanks

Samfox2, 2017-07-21:
I have the same problem.
What I did (as a dirty workaround) was putting the curl command in a loop and exit the loop if a file was written.

MrMontesa, 2017-07-21:
Thanks for info and good to know that its not limited to my local system. And yeah, for testing I did the same. But I do rely on a 3rd party application (openhab2) which uses the java curl library where this cant be done. So I'm keen on getting behind this issue.

MrMontesa, 2017-07-24:
I did a lot of testing ending up in tcpdumps and came to know that motioneye does not seem to send the image but sends an HTTP response of 200 - OK.
Since this is a new story I created #601 in order to analyze and troubleshoot in more detail. Would love to see some traction in the new ticket.

ccrisan, 2017-08-05:
This should be fixed with 0.36.1. Also the next version will have manual snapshot support using a button on the camera frame (added with https://github.com/ccrisan/motioneye/commit/55035be4624bad97e8fd262aa5b91aa196c373e2).

closed 2017-08-05 by ccrisan

jshep321, 2020-01-23:
Hi all,
Thought I'd share the "remote" snapshot button I found!
Motion has the native ability to trigger snapshot also.  In Motioneye the control port for this is on 7999.  The below curl command causes the "lastsnap.jpg" (and associated datestamped file) to be stored in the /var/lib/motioneye/Camera1 directory.

`curl -s -o /dev/null http://127.0.0.1:7999/0/action/snapshot #this causes a snapshot `

Then I can copy/link to that using my bash program...

kiteloopdesign, 2022-01-16:
For the life of me I can't find that snapshot button anywhere. Someone can kindly point where it is?

kiteloopdesign, 2022-01-16:
> Hi all, Thought I'd share the "remote" snapshot button I found! Motion has the native ability to remotely trigger a snapshot. In Motioneye the control port for this is on 7999. The below curl command causes the "lastsnap.jpg" (and associated datestamped file) to be stored in the /var/lib/motioneye/Camera1 directory.
> 
> `curl -s -o /dev/null http://127.0.0.1:7999/0/action/snapshot #this causes a snapshot `
> 
> Then I can copy/link to that using my bash program...

By the way, this one works for me, but implies ssh'ing into the Raspberry pi. Tried with my local IP instead of the loopback IP but that doesn't seem to work...

kiteloopdesign, 2022-01-16:
Finally found where this button is... Adding it here in case it can help someone

[image]

starbasessd, 2022-01-16:
@kiteloopdesign Or you can:
Settings, Still Images, Enable Manual Snapshots
and when you click in a frame, you get the camera button in the lower right of the picture:
[image]