motioneye-project/motioneye #97: layout columns prefs pannel is no longer available

John-Franck, 2016-01-10:
Hello,
I freshly installed Meye 0.30 from the Pip repository .
Two days ago I made an install from Meye, it was available a panel for the layout columns, but the last time I installed yesterday evening Meye this panel has disappeared.

How can I install it ?

thank you for your answer.

salutations

JFS

ccrisan, 2016-01-10:
I apologize but I don't understand what your problem is. Could you please try to explain it a bit clearer: what do you expect to have and you don't have. Maybe a screenshot would help.

John-Franck, 2016-01-10:
I'm sorry for this false problem.

Wanting to answer your question with a screenshot, I increased my window and the preference pane layout columns appeared ...
I'm a little green .. :-)

Your work is excellent.

Thank you for your attention.

JFS

ccrisan, 2016-01-10:
Ah, you're right. The layout columns option makes no sense for small screens and therefore it is only visible on screens larger than 1000 pixels :)

closed 2016-01-10 by ccrisan