motioneye-project/motioneye #98: Extra Motion Options Field

vdagamb, 2016-01-11:
Hello Crisan,
thanks for the great piece of software. I just set-up verion 0.30 on a fresh Jessie Raspberry2  and runs like hell. One question from my side: I noticed the field "Extra Motion Options" . To limit the detection area, I inserted "  area_detect 9  " (without quotes) and I feel it does not work. Moreover, I did not see it in motion.conf in my  ~/motioneye/conf directory. Anything else I should consider?

Thanks a lot,
Bernd

ccrisan, 2016-01-11:
You should see it in ~/motioneye/conf/thread-1.conf file. motion.conf contains the general (camera-independent) options.

vdagamb, 2016-01-12:
That really makes sense ... should have known by my own. Thanks, Bernd

closed 2016-01-12 by vdagamb

tekgeek1, 2016-09-18:
vdagamb - were you ever able to get the area_detect working?  I tried adding area_detect 3 to my thread file and noting happened.