motioneye-project/motioneye #99: No previws pics?

Castor123, 2016-01-11:
Hi,
in the captured video I get No thumbnails .

thanks

ccrisan, 2016-01-12:
Please attach your motioneye log file.

aleixpellicer, 2016-01-13:
I thought it was normal, I also don't get previews, neither can play the videos (only download them).

Here's my log:
[motion.txt](https://github.com/ccrisan/motioneye/files/89185/motion.txt)

ccrisan, 2016-01-13:
motioneye.log, please

aleixpellicer, 2016-01-13:
@ccrisan there's no motioneye.log, i start it on rc.local with this if it helps: ``sudo /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf &``, should I do it this way? ``sudo /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf > /var/log/motioneye.log &`` ?

Castor123, 2016-01-14:
Please look this...

 ERROR: failed to create movie preview for /var/lib/motioneye/2016-01-15/00-24-23.avi: Command 'ffmpeg -i "/var/lib/motioneye/2016-01-15/00-24-23.avi" -f mjpeg -vframes 1 -ss 4 -y /var/lib/motioneye/2016-01-15/00-24-23.avi.thumb' returned non-zero exit status 127

Castor123, 2016-01-14:
pi@raspberrypi:/var/lib/motioneye/2016-01-15 $ sudo ffmpeg -i "/var/lib/motioneye/2016-01-15/00-24-23.avi" -f mjpeg -vframes 1 -ss 4 -y /var/lib/motioneye/2016-01-15/00-24-23.avi.thumb
ffmpeg: error while loading shared libraries: libopencore-amrwb.so.0: cannot open shared object file: No such file or directory

ccrisan, 2016-01-15:
@Castor123 please follow the ffmpeg-related instructions from [this](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian#instructions) wiki page.

More precisely, make sure you run an up-to-date Raspbian Jessie, remove any existing ffmpeg/libav-related packages and install them as indicated.

aleixpellicer, 2016-01-15:
@ccrisan same problem here, got this error instead:

``ERROR: failed to create movie preview for /var/lib/motioneye/2016-01-15/21-18-02.avi: Command ´ffmpeg -i "/var/lib/motioneye/2016-01-15/21-18-02.avi" -f mjpeg -vframes 1 -ss 4 -y /var/lib/motioneye/2016-01-15/21-18-02.avi.thumb´ returned non-zero exit status 127``

ccrisan, 2016-01-15:
Do you run an up-to-date raspbian jessie?

aleixpellicer, 2016-01-15:
@ccrisan I run Wheezy and if updated means that I ran apt-get update lately, yes I did

ccrisan, 2016-01-15:
I can only help you with Jessie. With wheezy you're on your own.

closed 2016-01-15 by ccrisan

spupuz, 2023-07-18:
hello there i have same problem with docker install dev/edge version, there are no pictures for the recorder videos

zagrim, 2023-07-19:
@spupuz Please open new issue with all the details (logs, relevant config etc) instead of commenting on an over seven year old closed issue.