motioneye-project/motioneye #102: h264 encoding and viewing video in interface

eborned, 2016-01-14:
I have been experimenting with the extpipe feature of 'recent' motion builds. This enables motion to record into arbitrary formats, including h264 in an mp4 container. MotionEye processes them correctly, as can be seen on this screenshot:

[image]

I've seen older conversations on implemention 'online' video playback, but one of the requirements was such a HTML5 video format. 

Is there currently someone working on this feature? Or was an implementation based on the extpipe system already started, but stopped because of difficulties.

I can see a couple of required changes:
* Method to get the MP4 stream to the client. I've tested this with the download URL for the video, and that works (just place the download URL in the video src tag).
* Method to set the extpipe configuration from MotionEye. I think a couple of default options would be fine. Possibly for now only the actual enabling of extpipe for the stream would be enough.
* I don't know if the on_movie_end method is also fired when extpipe is used. If that's the process that creates the video thumbnails: that part works.

Anything else I'm missing?

eborned, 2016-01-14:
For reference, I'm using this extpipe configuration. I haven't tweaked it yet, but for now, it works:

```
use_extpipe on
extpipe mencoder -demuxer rawvideo -rawvideo w=640:h=480:i420 -ovc x264 -x264encopts bframes=4:frameref=1:subq=1:scenecut=-1:nob_adapt:threads=1:keyint=1000:8x8dct:vbv_bufsize=4000:crf=24:partitions=i8x8,i4x4:vbv_maxrate=800:no-chroma-me -vf denoise3d=16:12:48:4,pp=lb -of lavf -o %f.mp4 - -fps %fps
```

ccrisan, 2016-10-26:
Recent versions of motion support mp4/h264 "natively" (i.e. without `extpipe`). motionEye will allow choosing mp4 as a container with h264 as the codec.

closed 2016-10-26 by ccrisan