motioneye-project/motioneye #105: High CPU Utilisation - motion on RPi 2B

phibes4000, 2016-01-18:
I have 3 cams, using RTSP to access.  I am running...
  RPi 2B
  Ubuntu Mate 15.10
  Mr dave motion fork

With a single cam @ 1280*720, 4fps, and no motion detection, streaming, recording enabled (just 'Video Device'), motion hits ~100% CPU utilisation.  Enabling continuous recording pushes this to ~130%.

With 3 cams and continuous recording I get an average of about 230% CPU utilisation.

This surprises me because motion should not be doing any motion detection, just deserialising and encoding to file.

Is this expected, and if so 'tunable'?

I was running motioneyeos which seemed to have much lower utilisation.  Motioneyeos was great except I need to run other small programs and buildroot is too fiddly/restrictive.

On a related note, does the RPi support for hardware mpeg decoding (poss encoding?) offer any CPU offloading options?

mvanbreeden, 2016-01-19:
With a single cam on a rpi 1, I rarely get CPU usage above 30% when idling.
When actively looking at cam and viewing recorded Movies I het a load
average of up to 1.68. Running on raspbian

ccrisan, 2016-01-19:
Yes, "viewing" involves spawning an internal MJPG client that captures frames from motion and serves them to the browser. And it appears to be quite CPU hungry. I currently don't have any ideas on how to improve this situation.

phibes4000, 2016-01-30:
Seems like a motion issue, possibly a ubuntu dependency as I had great performance for the same Mr Dave build as used on the motioneyeos buildroot image.  As much as I really like the motioneye UI, 'm going to use ffmpeg directly and ditch motioneye as I don't really need motion detection.  Good luck!

closed 2016-01-30 by phibes4000