motioneye-project/motioneye #106: Webhook notifications failing with 400

drewdah, 2016-01-19:
First of all, thank you for the time you put in making this project, I was able to get a security camera for my front door working in a few hours with Raspberry Pi and the camera module (https://plus.google.com/+DrewLawton/posts/UAFsLYpxUfd)

One thing I haven't been able to get working is motion notifications via webhook. The URL I'm using is for a service called Pushover that sends push notifications to your devices whenever the url is hit. 

This is the URL (with my keys omitted):
"http://api.pushover.net/1/messages.json?token=_myapikey_&user=_myuserkey_&message=MotionEye"

Whenever it detects motion I see this in the motioneye.log:
2016-01-19 12:19:29: [webhook]    ERROR: failed to call webhook: HTTP Error 400: Bad Request

I tried sending a POST request manually to the URL and it works fine, I tried changing the different POST methods in the MotionEye config and unfortunately nothing worked. 

I was able to work around it by creating a little python script that makes the same request and used the "run a command" setting to execute the script, but I thought I'd let you know.

Using version 20160109 on a Raspberry Pi 2

ccrisan, 2016-01-20:
Hi,

Nice pics. Is the case IP65 or IP something? Could you provide a link to the product?

On to your actual problem now. I admit the webhook functionality hasn't been tested too much lately. Please attach your motioneye.log file with log_level = debug and containing the part where the webhook does its thing. Also please let me know what your webhook settings are (apart from the URL which you have already mentioned). Maybe a small screenshot would help.

drewdah, 2016-01-21:
The case I put the Pi2 into is this fake security camera http://www.harborfreight.com/outdoor-imitation-security-camera-61806.html that I found at a local hardware store. I brought the Pi with me after seeing it online and visually sized it up before purchasing. I had to remove some of the plastic screw posts on the inside with a dremel tool and solder power directly to the 5v and ground on the board, but it fits well and still has structural integrity from screws from the front as well as a twist on cap from the back.

As for my issue with webhooks, I'll grab some logs and a screenshot of my configuration and get back to you.

drewdah, 2016-01-24:
I think I may have figured out the issue here. 

In some of my testing the URL I was posting to had URL encoded characters in them (ex: &message=Motion%20Detected) and from the help tooltip it appears that you're using those as tokens for string replacement. The %20 appears to be replaced with the date according to the log file, so I removed the URL encoded character from the webhook URL I was testing with and it worked with the POST (query) HTTP method.

I'll probably keep using the command to run the python script as I'm not sure how you'd want to try and solve that other than using different replacement tokens for webhook URLS.

I've attached a motioneye.log snippet just in case its useful. Thanks again

[motioneye-webhook-log.txt](https://github.com/ccrisan/motioneye/files/102535/motioneye-webhook-log.txt)

ccrisan, 2016-01-25:
Yes, you're right, the %-escaped characters in the URL pose problems. I'll see what I can do about this.

ccrisan, 2016-07-03:
This should have been fixed a long time ago.

closed 2016-07-03 by ccrisan

00lex, 2020-08-01:
> This should have been fixed a long time ago.

definitiv not! :(