motioneye-project/motioneye #107: Motioneye Stream Freezes.

db414813, 2016-01-20:
So I have motioneye on ubuntu 14.04 and a fast mjpeg stream from an rp2 going to motioneye. Both have the latest motioneye/os. After a while the stream will freeze. Leaving just a still on the motioneye page. I restart the pi and nothing happens. I restart the motioneye pc and everything is good again. Any ideas?
[log.zip](https://github.com/ccrisan/motioneye/files/96762/log.zip)

ccrisan, 2016-01-20:
So let me see if I get your scenario: you have a RPI2 running motionEyeOS with Fast Network Camera enabled. On another machine you have an Ubuntu with motionEye. How have you added the RPI2 camera to your Ubuntu-based motionEye? As a Network Camera? as a Remote motionEye Camera or as a Simple MJPEG Camera?

db414813, 2016-01-20:
You are correct. It is as a network camera with motion detection.

db414813, 2016-01-20:
Also have a c920 web cam connected to Ubuntu machine.

db414813, 2016-01-20:
Like to say thank you for developing Motioneye. It's mighty cool.

ccrisan, 2016-01-20:
Actually this is the best scenario and I myself have been using motionEye(OS) this way for some time now. Here are some ideas for a further investigation:

 * send me all the logs from your motionEyeOS system
 * try keeping an open browser window with the streaming URL of your motionEyeOS and see if it also freezes when the motionEye frame freezes

db414813, 2016-01-23:
Ok, it froze again. I emailed you every log file I could find and a screen shot of the freeze. The second fps "1.9" is stuck at that while the other fps is still changing?

db414813, 2016-01-23:
Some more info. I used Mr.Daves motion and install ffmpeg via some ppa which I could dig up if you want me too. After installing ffmeg the guide I used also called for me to use apt get dist-uprade which I avoided because I thought it was a terrible idea haha.

ccrisan, 2016-01-23:
I replied to your mail saying that your logs are full of brownout warning messages. Please replace your PSU with a more powerful one and see if the problem persists.

closed 2016-07-03 by ccrisan