motioneye-project/motioneye #108: Raspbian - MotionEye

mastix, 2016-01-22:
Hey Crisan,

I've just installed MotionEye on my RPI 2 running Raspbian Jessie.

I uploaded the my config from motioneyeOS, I hope that does not create a problem.

Anyway, for some reason my email notifications do not work. In order to find out what the problem is, I've checked the /var/log folder and it seems that it does NOT contain a motioneye.log file, just the motion.log. In the motioneye.conf /var/log is set as log folder.

Any idea what's causing this?

Regards and thanks for that great product,

Sascha

ccrisan, 2016-01-22:
The backup/restore functionality is not compatible between motionEye and motionEyeOS. Jessie uses systemd which relies on its own journaling/logging mechanism instead of files in `/var/log/`. Use `journactl -u motioneye` to see those logs.

mastix, 2016-01-22:
Ok, that's what I thought... :(

Is there a way to clean up my motioneye conf completely? Does a re-installation remove all imported configs?

mastix, 2016-01-22:
Ok, fixed it by just pointing to a new config directory. Everything works now as expected.

Btw: I had to install pytz (pip install pytz) to make it work. Couldn't find that in the docs.

ccrisan, 2016-01-22:
`pytz` is not a requirement. It is only needed when you want to control your timezone via the UI, which is unusual for a normal motionEye installation.

closed 2016-01-22 by ccrisan