motioneye-project/motioneye #109: Unable to open video device / rtsp

Dobi29, 2016-01-23:
Hello,
i have installed motioneye 0.30. My picam works fine. Now i want to add a network camera.
[image]
..but i get no stream...
[image]
Using VLC with rtsp://192.168.178.26 i get the stream of the camera.
Thanks for your help.

ektus, 2016-01-23:
Do you have motion-mrdave installed? What's in the logfiles? Does the camera require authentication? 

Also, in my experience with motioneye rtsp does need a full path (including port number) to the stream, not merely the device IP address. For my QCam this reads "rtsp://192.168.0.12:554/h264Preview_01_sub"

Regards
Ektus.

Dobi29, 2016-01-24:
..thx for the reply. I try it with full path and authentication. The same. Now i want to install motion-mrdave for raspbian  --> https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian. 

note 2: A prebuilt version of Mr Dave's motion for Raspbian can be downloaded from here. Just download it and copy it to /usr/local/bin/motion

i cant' find the directory "motion"   and there is no solution in "google"....

How can install mrdave under respbian ? 

Regards Klaus

ccrisan, 2016-01-24:
There's no motion folder and there shouldn't be one. Copy your motion-mrdave binary to /usr/local/bin and rename it to motion (or shorter, copy it to /usr/local/bin/motion).

Dobi29, 2016-01-24:
Thx. Then restart the pi or going to step 3 ? How can i check using mrdave ?

Dobi29, 2016-01-25:
that's my log-File. My Cam : SN-IPC 5033SW-WI-EU

[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "rtsp_uses_tcp"
[0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.31/)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.178.31/)
[2] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
[2] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.31/)
[2] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.178.31/)
[2] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.31/)
[2] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.178.31/)
[2] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.31/)
[2] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.178.31/)
[2] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.31/)
[2] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.178.31/)

ektus, 2016-01-25:
That camera should be from Sinocam, and according to 
https://www.ispyconnect.com/man.aspx?n=Sinocam I'd try and set the address in motion to rtsp://190.168.178.31:554//ucast/11 or maybe rtsp://190.168.178.31:554//11

You may also want to check if vlc shows a complete path.

Dobi29, 2016-01-25:
i was on this side yesterday and tried all adresses...i don't konw if have installed mrdave correct. how can i check this ?
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "rtsp_uses_tcp"
[0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
i copied mrdave into  /usr/local/bin. thats all ?

ektus, 2016-01-25:
I don't have access to my RPi during the week, so can't answer properly. But normally copying the binary is enough, and the version string doesn't look too wrong either, at least some date in 2014 could be correct. As a last resort, you could try and compile yourself, but first I'd try and investigate which complete url leads to the desired stream. In case the camera is ONVIF capable, you could try the ONVIF Device Manager to view it's properties. http://sourceforge.net/projects/onvifdm/

Dobi29, 2016-01-25:
[image]

next step , only one "/" 

[image]

the same

[image]

Dobi29, 2016-01-25:
log-file

[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-3.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "rtsp_uses_tcp"
[0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[3] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.31:554/ucast/11)
[3] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.178.31:554/ucast/11)
[3] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
[3] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[3] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.31:554/ucast/11)
[3] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.178.31:554/ucast/11)
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[3] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[3] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.31:554/ucast/11)
[3] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.178.31:554/ucast/11)
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

Dobi29, 2016-01-28:
i made a second installion with motioneyeos and there i can get the cam with rtsp://ipadresse:554/ucast/11

ccrisan, 2016-01-29:
Most likely you weren't using motion-mrdave. I'll close the issue now.

closed 2016-01-29 by ccrisan

DrCord, 2016-09-11:
I am still getting this problem with a Wansview 720P network camera. It works perfectly fine via VLC and through Onvif device manager with the url: rtsp://192.168.1.113:554/live/ch0 but shows the grey screen as show via the above screenshots...

DrCord, 2016-09-12:
This was caused by using a "+" in the password for the camera, apparently that character cannot be sent correctly in the authentication phase from motioneye...?

blindrain, 2018-06-22:
I have this problem on x86 and x64 ubuntu systems  following your directions to install motioneye on these systems .  I get 401 unauthorised errors in the logs but if I use straight ffmpeg to record the video from these systems I'm able to record the video.    this is an rtsp feed. and I have no special characters in my password.

blindrain, 2018-06-23:
how do I use ffmpeg to restream this so it works in motion eye. 

 ffmpeg -i rtsp://username:password@10.24.23.15:80/videoMain -acodec copy -vcodec copy -f h264 udp://0.0.0.0:1234/videoMain

Doesn't seem to work.