motioneye-project/motioneye #110: RTSP

chuck78, 2016-01-23:
Hello, 
When I try to add a RTSP camera to the system it asks for a usename and password. A dialog box, like the one when you log in, pops up asking for a username and password. but no matter what username and password I try it keeps popping up. Is this dialog box for the system admin u&p or for the RTSP u&p??

Thanks in advance

ektus, 2016-01-23:
That's for the RTSP u&p. I was able to provide these (IIRC) by first switching the camera type to network, then entering username and password, and then entering the network address. OTOH, in my camera those credentials are the same for admin and RTSP.

chuck78, 2016-01-24:
ok cool thanks! now i am getting the 401 error when I try to connect. :(

motioneyeosuser, 2016-03-29:
Hi, i add a HiKam S5 Cam (RTSP) to my Motioneye Raspi. I get the following error message: Unable to open Video Device. Has anyone experience with the camera or some tips? Resolution is 1280 x 720. Greets

motioneyeosuser, 2016-03-29:
Ok, i have found the error. the camera needs no authentication! :-)
Url: 
rtsp://camera_ip:554/onvif1

LeonSerri, 2016-07-02:
Dear All,
I have the same issue. When I try to add an IP camera using rtsp 554 port, motioneye recognizes the ip camera (sricam an003) as HIPcam/V100R003 VodServer/1.0RTSP/UDP Camera. But after that, a grey screen appears with the text "unable to open video device".

The log is as follow:
0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.64:554/ovif1)
[1] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.1.64:554/ovif1)
[1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
[1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera

Any suggestions to solve this problem? 

Thank you

ccrisan, 2016-07-02:
@LeonSerri what mition version do you use? What does `motion -h` say?

LeonSerri, 2016-07-02:
motion Version 3.2.12+git20140228

ccrisan, 2016-07-02:
@LeonSerri please use Mr Dave's version. [This](https://github.com/ccrisan/motioneye/wiki/Compiling-Motion) will walk you through compiling it.

LeonSerri, 2016-07-02:
I forgot to say you that I am using a Raspberry Pi B+. Is it compatible with this hardware?

ccrisan, 2016-07-02:
See [these instructuions](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian#instructions). You'll see a link to a precompiled Mr Dave's motion for the RPi.

LeonSerri, 2016-07-02:
I have followed the links that you have indicated and after motion -h command, RPi returns:

motion Version Unofficial-Git-UNKNOWN, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavassen

Is it correct?

ccrisan, 2016-07-02:
Sounds good. Does it work now?

LeonSerri, 2016-07-02:
Unfortunately no......same behaviour.

Suggestions?

LeonSerri, 2016-07-02:
No.

This is the log:
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.64:554/ovif1)
[1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://admin:admin@192.168.1.64:554/ovif1): Invalid data found when processing i$
[1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
[1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera

ccrisan, 2016-07-02:
In that case I won't be able to help you. It's either a problem with the stream or it's a bug in motion.

closed 2016-07-02 by ccrisan

LeonSerri, 2016-07-02:
I have found an old issue, in which you have suggested to use the full path for connection.
https://github.com/ccrisan/motioneye/issues/109

This works for me!!
Thank you again for your  support!

ccrisan, 2016-07-02:
I thought you were already using the full path. Could you please list the both variants here so that things be a bit clearer for everyone?

LeonSerri, 2016-07-02:
I have used this full path for connection:

rtsp://myipaddress:554//ucast/11
I have inserted the credentials (user and password) and selected the TCP connection:
HIPcam/V100R003 VodServer/1.0RTSP/UDP

dbk3r, 2017-09-17:
maybe a bit late, but use option "rtsp_uses_tcp off" in motion.conf to force udp tranport. 
this helps in my case.