motioneye-project/motioneye #111: Use "SMTP Account" or add separate "From email" field to "Send An Email" block

PaulAnnekov, 2016-01-25:
As for now it looks like email notifications are focused on sending email to yourself:
https://github.com/ccrisan/motioneye/blob/b409d9faa0bda143f4198f434719227041d0ae25/motioneye/sendmail.py#L59
`_from = 'motionEye on %s <%s>' % (socket.gethostname(), to[0])` 

`to` in `from`, lol. It's ok when you send email from you@gmail.com to you@gmail.com using gmail's smtp server. But it's wrong for the case when you have your send-only email servers (e.g `exim4`) on each server machine with motioneye and want to collect all emails in gmail account. In this case you need to make some hack to provide correct `From` field to gmail:
`Email addresses`: you@gmail.com, motioneye@first.server
`SMTP server`: 127.0.0.1
`SMTP port`: 25

`From` field will be correct, but it will send a message to two emails, that I don't need.
Suggestion: use `SMTP Account` input as email part of `From` field. It's not always full email address, but we can try to add SMTP Server if it's not an address. Or add separate "From email" field.

ccrisan, 2016-01-26:
This is the way it was a few versions ago. Unfortunately some SMTP servers (I don't recall which exactly) didn't like it that way. I believe the correct solution here is to provide a separate UI option for the `From` field.

The reason why I didn't implement it this way yet is the backwards compatibility issue. More precisely, the number of (positional) arguments passed to the sendmail.py script would increase by one and practically you'd need to reconfigure your settings upon upgrade.

PaulAnnekov, 2016-01-26:
I thought that you have mechanism of config migrations from version to version.

ccrisan, 2016-02-21:
The "From" field is now configurable in the UI. The change will be available starting with 0.31.

closed 2016-02-21 by ccrisan

PaulAnnekov, 2016-02-21:
@ccrisan good job, 10x :)