motioneye-project/motioneye #112: HTTP error 500

L3o-pold, 2016-01-26:
After few days without rebooting, I can access to the web interface but without any JS or CSS files.
Firebug display an error 500 for all files excepted the home page.

[image]

The bug disappear after a reboot. 
Here is the log, it's a similar message for every ressource.

```bash
2016-01-26 21:18:09: [motioneye]    ERROR: 500 GET /static/img/motioneye-logo.svg (192.168.0.x) 4.29ms
2016-01-26 21:18:09: [motioneye]    ERROR: Uncaught exception GET /static/js/ui.js (192.168.0.x)
HTTPServerRequest(protocol='http', host='192.168.0.x:xxxx', method='GET', uri='/static/js/ui.js', version='HTTP/1.1', remote_ip='192.168.0.x', headers={'Accept-Language': 'fr-FR,fr;q=0.8,en-US;q=0.6,en;q=0.4,ru;q=0.2', 'Accept-Encoding': 'gzip, deflate, sdch', 'Host': '192.168.0.x:xxxx', 'Accept': '*/*', 'User-Agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X x_x_x) AppleWebKit/xxx.xx (KHTML, like Gecko) Chrome/x.x.xxxx.xxx Safari/xxx.xx', 'Connection': 'keep-alive', 'Referer': 'http://192.168.0.x:xxxx/', 'Pragma': 'no-cache', 'Cache-Control': 'no-cache'})
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1381, in _execute
TypeError: decode_argument() takes exactly 3 arguments (2 given)
```

ccrisan, 2016-01-27:
I couldn't tell what the heck is going on with that error. It looks like a messed up tornado installation or something like that. What's your OS? What version do you run? Have you tweaked your motioneye.conf?

L3o-pold, 2016-01-27:
Motioneye OS, latest version without any tweaking. 
I understand, that it's hard to investigate on this case.
Is there anything I can do to help you to debug this if this bug appears again?

ccrisan, 2016-01-27:
Ah, I thought it was a plain motionEye. I suspect a corrupt installation then. Well, if you could set up remote SSH access (and if that still works with your system), I'd be happy to try and identify the problem remotely.

L3o-pold, 2016-01-27:
Ok, my installation was several months ago and was updated.
Like I said, the bug disappear after a manual reboot so if it appears again, I will let you try to debug it via SSH.

closed 2016-01-27 by L3o-pold