motioneye-project/motioneye #113: What does it mean the upper right integer

chfakht, 2016-01-28:
Hi , I’m using the latest rapbian jessie with motionEye , it'w working perfectly and the image are beign uploaded to google Drive :) 
**I want to know what does it mean  the upper right integer that appears on the camera capture (1800 , 80 , 58 ... )** i think it's related to motion Detection but i don't understand what it really mean :large_orange_diamond:

ccrisan, 2016-01-29:
That's the number of changed pixels.

closed 2016-01-29 by ccrisan

chfakht, 2016-01-29:
Please  what is the algorithm used for motionDetection ?

mvanbreeden, 2016-01-30:
It's in Motion by MrDave, which is used by MotionEye. Google it, you'll see that motion detection is done by comparing pictures and pixel change.

chfakht, 2016-01-30:
@mvanbreeden  thanks i have  found a link here 
 [http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionTechnology](url)
but it's not complete as i think , could you tell me where to find the source code for it.
many  thank

mvanbreeden, 2016-01-30:
Sure @chfakht, it is the link that Calin has referenced on his github
repos: https://github.com/Mr-Dave/motion

chfakht, 2016-01-30:
@mvanbreeden ok i have found it (1000 lines of code :mag: )