motioneye-project/motioneye #115: Ability to play recorded videos in browser

PaulAnnekov, 2016-01-31:
Most browsers support playing video using `<video>` tag ([list of supported codecs/containers for browsers](https://developer.mozilla.org/en-US/docs/Web/HTML/Supported_media_formats#Browser_compatibility)). Mr Dave's `motion` [supports some formats](https://github.com/Mr-Dave/motion/blob/master/ffmpeg.c#L148) that can be played in browser (mp4, ogg, mkv).

You just need to detect the container of video from a list (it will be better to detect codec too), pass it to client side and show `Play` button in enabled/disabled state for each video. You can check if browser supports this container/codec using [canPlayType](http://stackoverflow.com/a/8134378/782599) method. Disabled state goes with `title` attribute with message: "Codec/container does not supported by your browser". On `Play` click it will show some html5 player like [Video.js](http://videojs.com/).

jeff-h, 2016-08-08:
I'd love this too. You sound like you know what's what in terms of the code on this. Any chance you could put a PR together for it?

PaulAnnekov, 2016-08-10:
Just checked the sources. Looks not so difficult. Will try.

george-vieira, 2017-02-19:
I've read #218 and unless I misread it too quickly, did this get merged as I just checked out latest showing up as version 0.35.2 and doesn't seem to be there?

Mrnt, 2018-03-15:
@george-vieira it got merged into the dev branch

ccrisan, 2018-06-20:
Closing as the functionality is now present in the latest version.

closed 2018-06-20 by ccrisan