motioneye-project/motioneye #116: Menu Options not appearing

TheNore, 2016-01-31:
I am sorry if this has been covered before, but I ham having an issue on a new install on a raspberry pi 2 version 20160109. I am running a Raspberry Pi Camera, and I can get video and cam settings with motioneyeOS but i can get no menus with any configuration for recording or motion detection, I assume it is because I do not have the movie recording option turned on, but I can not find any reference to this in any of the config files. I have attached my logs as well. Please let me know if i can provide any additional information to resolve this problem. 

[dmesg.txt](https://github.com/ccrisan/motioneye/files/111358/dmesg.txt)
[motioneye.txt](https://github.com/ccrisan/motioneye/files/111351/motioneye.txt)
[messages.txt](https://github.com/ccrisan/motioneye/files/111352/messages.txt)
[motioneye (1).txt](https://github.com/ccrisan/motioneye/files/111353/motioneye.1.txt)
[raspimjpeg.txt](https://github.com/ccrisan/motioneye/files/111354/raspimjpeg.txt)
[boot.txt](https://github.com/ccrisan/motioneye/files/111355/boot.txt)

ccrisan, 2016-02-01:
Your log files look alright. Please attach a screenshot that further explains your problem.

TheNore, 2016-02-01:
Sorry I was able to fix this by disabling fast network camera

closed 2016-02-01 by ccrisan