motioneye-project/motioneye #117: Motioneye SLOW thumbnails since upgrade to 20160109

tomskidoodle, 2016-02-03:
Since last OS upgrade (20160109) MotionEye is taking an age to download thumbnails (circa 30 seconds each)  When you have hundreds of clips a day this almost unusable.  Prior to this upgrade it was a super smooth system...

ccrisan, 2016-02-03:
I doubt it has anything to do with the update. Please attach your logs to see if there's anything suspicious going on.

tomskidoodle, 2016-02-03:
Logs attached as requested, thanks.  There seems to be a lot of errors in the motioneye.log file.

Tom

[boot.txt](https://github.com/ccrisan/motioneye/files/116205/boot.txt)
[dmesg.txt](https://github.com/ccrisan/motioneye/files/116201/dmesg.txt)
[messages.txt](https://github.com/ccrisan/motioneye/files/116202/messages.txt)
[motion.txt](https://github.com/ccrisan/motioneye/files/116204/motion.txt)
[motioneye.txt](https://github.com/ccrisan/motioneye/files/116203/motioneye.txt)

ccrisan, 2016-02-04:
You have either connected your motionEyeOS directly to the public internet or configured port forwarding on 80. Please use a different port (e.g. 8888) or otherwise your motionEyeOS will be flooded with unwanted requests.

tomskidoodle, 2016-02-04:
Hi

Yes I have port forwarding on 80 however this was in place before the update and all worked fine.  Immediately after the update the problems started.  I waited a few days in case it was a common issue that would be sorted.

I have now changed to 8888 but after performing the required reboot I have totally lost the system and will have to reinstall!

ccrisan, 2016-02-04:
You shouldn't have changed the port on motionEyeOS side, but rather on your router, in the port forwarding configuration. The incoming 8888 port should be forwarded to port 80. motionEyeOS shouldn't require any change or reboot (let alone a reinstall).

closed 2016-02-04 by ccrisan

tomskidoodle, 2016-02-04:
Hi

The port forwarding on my router only allows same port to port, i.e port 80
to port 80 but not port 8888 to port 80, hence having to change port on the
Pi.  I got it running again without reinstall on port 80 and have, for now,
turned off the external port forwarding.

This seems to have made no difference though to the thumbnail download
delay.

Thanks