motioneye-project/motioneye #118: RTSP-No connection

andre1000, 2016-02-03:
I have a problem with the connection of a IP-cam via rtsp protocol.
The connection with VLC with 
rtsp://192.168.x.x:554/onvif1
works fine.

When I use the same address in motion I become the following message:
- netcam_start: Network Camera thread starting... for url (rtsp://192.168.x.x:554/onvif1)
- netcam_rtsp_open_context: Using tcp transport
- netcam_rtsp_open_context: unable to open input(rtsp://192.168.x.x:554/onvif1): Invalid data found when processing input

A other rtsp cam works fine with motion
Thanks,
André

ccrisan, 2016-02-04:
This is a motion-related issue and cannot be fixed or worked around by motionEye.

closed 2016-02-04 by ccrisan

andsk8, 2017-03-19:
I have the same problem. I have motion working with one cam perfectly. TP-Link NC250. As soon as I change the ip to the other cam, and I receive all the same errors than you. Both cams are the same, and both work on vlc with the same kind of url. rtsp://user:pass@host:554/h264_hd.sdp
Any advice?

asquelt, 2017-03-21:
same problem here: rtsp://192.168.111.218:554/onvif1
works fine under vlc.
returns "Invalid data found when processing input" under motion.
it's most likely some missing library.
camera is Digoo DG M1Q.

ccrisan, 2017-03-22:
@asquelt see [this](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) for details on how to report this to motion.

dbk3r, 2017-09-17:
add 
rtsp_uses_tcp off

in motion.conf to force udp transport

jeanbruder, 2018-02-10:
@dbk3r : Thx so much ! This solved my issue with motion and my SRICAM SP009 !

gusarg81, 2018-11-18:
> add
> rtsp_uses_tcp off
> 
> in motion.conf to force udp transport

Solved for me to for SRICAM SP017! Thanks!!

btpys, 2019-06-06:
> add
> rtsp_uses_tcp off
> 
> in motion.conf to force udp transport

Thx! solved it for a Besder (Yoosee 6024PW-JW201) as this cam only works with UDP.

1. Find the cam config file, it should be something like this: /data/etc/thread-X.config
2. add: rtsp_uses_tcp off
3. Reboot SO.

bianco-ciro, 2019-08-21:
> > add
> > rtsp_uses_tcp off
> > in motion.conf to force udp transport
> 
> Solved for me to for SRICAM SP017! Thanks!! In my case I just added this param to this IPCAM (and not entire motion config).

How you have do that ?

gusarg81, 2019-08-21:
> > > add
> > > rtsp_uses_tcp off
> > > in motion.conf to force udp transport
> > 
> > 
> > Solved for me to for SRICAM SP017! Thanks!! In my case I just added this param to this IPCAM (and not entire motion config).
> 
> How you have do that ?

You have to add that line to the ipcam config.