motioneye-project/motioneye #121: Passwords dumped into /data/etc/motion.conf in plain txt?

gittrdonebrad, 2016-02-11:
Seriously?  My old version of motion running on another pi doesn't require this.  I was trying to find info on the frame rate dimmer slider that was apparently added to motioneye and checked the motion.conf hoping for comments...  What I did find instead is NOT COOL!
```
cat /data/etc/motion.conf
# @normal_password BLAHBLAHBLAHREMOVED
# @enabled on
# @show_advanced on
# @admin_username admin
# @admin_password BLAHBLAHBLAHREMOVED
# @normal_username user
###SNIP
```

ccrisan, 2016-02-12:
Given your findings, I wouldn't use this piece of software if I were you. Thinking that someone having my SSH credentials, having access to my local network, having Linux experience and knowing about this serious security issue would find out what my motionEye password is, wouldn't let me sleep at night.

rickerd, 2016-04-28:
Know how Linux work isn't that hard. A script kiddy is able to find this information yes.

closed 2016-07-03 by ccrisan