motioneye-project/motioneye #122: Network camera PTZ control

Snippo, 2016-02-14:
I just started using MotionEye after using a Foscam web interface for years but I'm missing one feature: PTZ controls.
Most of the network cameras (all of which I own, Foscams and Wanscam) are controlled like this:
http://ip:port/decoder_control.cgi?user=*user*&pwd=*pwd*&command=1|2|3|4
In which 1, 2, 3 and 4 are commands for up, down, right, left (not sure if the values correspond to the commands in that order). To stop the movement it sends an empty command.
It is pretty simple to add some buttons like this to a HTML file, but I'm not sure how to add some extra buttons to the interface of MotionEye.
Do you think PTZ control can be added?

ccrisan, 2016-02-14:
This is definitely not on the list, as things can be terribly different from one camera to another when it comes to pan/tilt controls. However, if you really need those options in your UI, you could do a workaround using [Action Buttons](https://github.com/ccrisan/motioneye/wiki/Action-Buttons). I know it's far from ideal but could do the job, if you're willing to write some minimal scripts.

Snippo, 2016-02-14:
I'll give it a look. Thanks.
The pan/tilt controls can indeed be terribly different. Fortunately newer cameras often support ONVIF which should all have equal PTZ control. Maybe implementing ONVIF PTZ support is an idea for the future?

I'll try making something work using the Action Buttons. If I get something decent to work I'll post it here.

Snippo, 2016-02-18:
I managed to hack something together using the action buttons. I created the action button files using the following command:
curl "http://ip:port/decoder_control.cgi?user=user&pwd=password&command=0&onestep=1"
(commands are 0, 2, 4 and 6 for up, down, left and right).
Works fine. If I find some spare time I'll try to add extra action buttons for PTZ controls instead of 'hacking' the default action buttons. For now I just replaced the icons of lock, unlock, light_on and light_off with up, down, left and right...

Puzzlr, 2016-02-18:
About the same thing here:

For a HD Foscam camera (for instance a FI9831P) the following command lets it move towards a named preset. 

curl -s "http://cam_IP:88/cgi-bin/CGIProxy.fcgi?cmd=ptzGotoPresetPoint&name=[Preset_Name]&usr=[username]w&pwd=[password]"

I used this to put two executable files (lock_1 and unlock_1) in the /etc/motioneye directory to make the camera pan to the two preset positions that I use most: a pefectly working solution. 

As per the remarks in the Wiki page:
do not use an executable file with an extension, but above that: a link-file linked to a file with an extension does not work either! 
(I thought to do a smart thing by doing "ln -s ./script_name.sh lock_1" ).

Puzzlr, 2016-02-18:
Would it be possible to create slots in the overlay button code for some "custom" buttons?

blackbirdstreet, 2016-11-23:
Hey Puzzlr! Your post wass very helpful to me. But I still have some problems using the action buttons.
I try to call a URL (http://[IP]/decoder_control.cgi?command=33&user=[user]&pwd=[password]) via an action button but cannot manage the button to be displayed... I created a file called preset1_2 in the folder /etc/motioneye Then I made the file executable (chmod +x /etc/motioneye/preset1_2). The I restarted the motioneye service but there is no button displaying as an overlay in the web frontend... Any help is apreciated! Thank you!

i007laser, 2017-06-29:
Probably a bit late for you blackbird, but just in case others have the same issue, the linked wiki page defines the action button file should be created in /etc/eyemotion/ 
After doing this nothing was displayed for me either, it was then I found a reference to the correct location (on a default install) is /data/etc/

Hope this saves others 5 minutes of frustration lol!

Harvie, 2017-09-30:
If there's standard on PTZ it should be quite easy to write bash script that will reside eg. in `/usr/share/motioneye` and will be symlinked for all cameras and actions that should use given PTZ standard (you can have multiple scripts for multiple standarts).

Then you symlink such script to multiple actions, eg. to enable onvif zoom in on camera 1 you will simply do:

`ln -s /usr/share/motioneye/onvif_ptz_driver /etc/motioneye/zoom_in_1`

the `onvif_ptz_driver` will then read `$0` variable (containing name of symlink called) to figure for which cam and which action it was called and will be even able to parse ip address of cam and login credentials from `/etc/motioneye/thread-1.conf`.

This is completely unobtrussive to motioneye architecture and very easy to configure (once somebody will write such onvif driver). Once it will be tested it might be even possible to create/delete these symlinks automatically if user will click on "enable onvif support" (pan+tilt, zoom and focus can be enabled separately depending on capabilities of camera). Also it should be checked if there's already such script and if it's really symlink to /usr/share/motioneye/*, so it does not accidentaly overwrite other custom scripts or symlinks that we might have configured manualy.

giuse320, 2018-05-17:
Mine IP cam floureon managed to make it work like that:

#!/bin/bash
# move right 1 sec then stop
curl --user  admin:password --user-agent "admin" --data "command=4&ZFSpeed=0&PTSpeed=0&panSpeed=1&tiltSpeed=1&focusSpeed=1&zoomSpeed=1" http://192.168.X.X:yy/form/setPTZCfg
sleep 1
curl --user  admin:password --user-agent "admin" --data "command=0&ZFSpeed=0&PTSpeed=0&panSpeed=1&tiltSpeed=1&focusSpeed=1&zoomSpeed=1" http://192.168.X.X:yy/form/setPTZCfg

sroettgermann, 2018-05-18:
Hey @giuse320, may I ask what specific kind of camera you are using?

alexmarsaudon, 2018-08-01:
Alright, this page started me down a rabbit hole that resulted in working PTZ buttons for my Amcrest IP2M-841W . In theory, this same approach should work for any modern Amcrest PTZ camera.
In this example, my camera's ID is "1". These files were placed in /etc/motioneye and made executable 
`chmod +x up_1`. 
Update the username ("admin" below) and password ("Censored" below) to fit your needs.
These were based off https://wiki.zoneminder.com/Amcrest and this user's post: https://community.home-assistant.io/t/command-line-amcrest-camera-help/25041/8 

File "preset1_1":
`#!/bin/bash`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=start&channel=0&code=GotoPreset&arg1=0&arg2=1&arg3=0&arg4=0"`

File "preset2_1":
`#!/bin/bash`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=start&channel=0&code=GotoPreset&arg1=0&arg2=2&arg3=0&arg4=0"`

File "preset3_1":
`#!/bin/bash`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=start&channel=0&code=GotoPreset&arg1=0&arg2=3&arg3=0&arg4=0"`

File "preset4_1":
`#!/bin/bash`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=start&channel=0&code=GotoPreset&arg1=0&arg2=4&arg3=0&arg4=0"`

While I find the presets to be the most effective and useful for PTZ, I also wanted navigation. The camera, by default, will continue moving in a direction until it receives a "stop" command - I introduced a simple sleep step followed by a stop step to allow this to increment movement.

File "up_1":
`#!/bin/bash`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=start&code=Up&channel=0&arg1=1&arg2=1&arg3=0"`
`sleep 1`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=0&arg3=0&arg4=0"`

File "down_1":
`#!/bin/bash`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=start&code=Down&channel=0&arg1=1&arg2=1&arg3=0"`
`sleep 1`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=0&arg3=0&arg4=0"`

File "left_1":
`#!/bin/bash`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=start&code=Left&channel=0&arg1=1&arg2=1&arg3=0"`
`sleep 1`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=0&arg3=0&arg4=0"`

File "right_1":
`#!/bin/bash`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=start&code=Right&channel=0&arg1=1&arg2=1&arg3=0"`
`sleep 1`
`curl --digest -u "admin:CENSORED" "http://10.10.110.14/cgi-bin/ptz.cgi?action=stop&code=Right&channel=0&arg1=0&arg2=0&arg3=0&arg4=0"`

Hope this saves someone some time. Now to get this working on a Hikvision DS-2DE3304W-DE !

McSpitz, 2018-10-20:
> Mine IP cam floureon managed to make it work like that:
> 
> #!/bin/bash
> 
> curl --user admin:password --user-agent "admin" --data "command=4&ZFSpeed=0&PTSpeed=0&panSpeed=1&tiltSpeed=1&focusSpeed=1&zoomSpeed=1" http://192.168.X.X:yy/form/setPTZCfg
> sleep 1
> curl --user admin:password --user-agent "admin" --data "command=0&ZFSpeed=0&PTSpeed=0&panSpeed=1&tiltSpeed=1&focusSpeed=1&zoomSpeed=1" http://192.168.X.X:yy/form/setPTZCfg

Great! It is working! Do you have an idea how I can say "go to preset position 4" or something like that?

odedgil, 2020-11-10:
+1 for onvif ptz controls

JamBalaya56562, 2026-11-13:
motionEye has action buttons for this: executable `up_<camera id>`, `down_…`, `left_…`, `right_…`, `zoom_in_…`, `zoom_out_…` and `preset1_…` to `preset9_…` files in the configuration folder show as buttons on the camera frame ([Action Buttons](https://github.com/motioneye-project/motioneye/wiki/Action-Buttons)), and the scripts above show how to call various cameras from them. Built-in ONVIF control is not planned, as @ccrisan said at the start. Closing; please reopen if something is missing.

closed 2026-11-13 by JamBalaya56562