motioneye-project/motioneye #124: IP-Camera as source

kurazli, 2016-02-15:
what kind of ip camera do you need to use it with motioneye?
I have two raspberry pies (old ones) with the csi cameraboard and i like to use them with a motion eye hub, because it has more power to calculate the motion etc.
however i can not add the cameras to my motioneye hub as ip cameras, only as remote motioneye cameras.
this might not be the right place here (sorry), but does anyone know how to solve this setup?

ccrisan, 2016-02-15:
You probably want to use motionEyeOS on your RPi and switch on the Fast Network Camera option. Either way, you can add them as IP cameras using port 8081.

kurazli, 2016-02-16:
Thank you for the reply.
Since the RPis are at a remote place (800+ km) I am kinda bound to Raspbian for know.
I will try adding them with Port 8081 in the meantime.

kurazli, 2016-02-18:
Maybe I should rephrase:
My setup is two RPi (first generation with 512mb of ram) in a lan *somewhere* far away running raspbian with the pi camera board.
Running motioneye on them generates a rather high load so i wanted to offload the motion detection to another server on the internet.
Unfortunately I can not use rtsp (network camera), since it is kind of impossible to get that through nat and it is even worse with two cams on the sam lan..
I then tried to add the camers to the server with mjpeg, but then I do not have the option for motion detection. Is motion detection not supported with mjpeg cameras?
So my only options would be using a ip camera (not possible because of rtsp and nat not working well together) or remote motioneye camera.
if I did get it right remote motioneye camera is really just a remote motioneye camera where all the work still happens on the pi itself and not the motioneye instance on the server.
is there any other way to make that setup happen like I described above and have my server calculate all the motion while using the remote RPis as video source?
Any hint would be appreciated

ccrisan, 2016-02-18:
If you added your cameras as Network Camera then motion detection should be available. If you added your cameras as Simple MJPEG Camera, then no, motion detection won't be available.

kurazli, 2016-02-18:
Thank you very much, that works indeed!

closed 2016-02-18 by ccrisan