motioneye-project/motioneye #125: WIFI Driver Problem

WanTanFao, 2016-02-16:
Hi,

a friend of mine is running your OS on his Pi to use it as a security camera. However, we have a problem with the driver. He's using Edimax 7811un USB WIFI Dongle which, with the currently loaded driver, is not able to work in n-Mode and therefore gets problems transferring videostreams (it hangs, then the connection breaks). Driver rtl8192cu is loaded but not used by the device. Why? And how may it be changed so the device gets a driver loaded that supports full n-Mode (or mixed mode, which is also not working on a TP-Link Archer C7) and with it the full bandwidth?!

Fabian

ccrisan, 2016-02-16:
The adapter you mentioned uses the driver 8192cu which is currently used by Raspbian as well. As far as I know, it works in n mode without a problem. How do you know it doesn't use 802.11n?

WanTanFao, 2016-02-16:
The adapter can't run with Archer C7's "Mixed Mode (b/g/n)". It chooses the lowest possible bandwidth when connecting to the AP in that Mode, which is, as I assume, a problem of the driver. Since the adapter can't run 5GHz (where we can set n-Mode only), this is a huge problem. And since "lsmod" shows 8192cu as not used anywhere, I suppose it's a driver issue with the legacy one that apparently has been loaded.

ccrisan, 2016-02-16:
What do you refer to as "legacy" driver? If lsmod doesn't list the 8192cu module, then your adapter might not be detected. I myself am using many Edimax 7811Un adapters and haven't had any real problems yet.

WanTanFao, 2016-02-16:
I'm not saying it's not loaded. It IS loaded but not used. See these links regarding the problem please (it's a kernel issue that has been fixed in a way newer Kernel than you are using): 

https://github.com/pvaret/rtl8192cu-fixes
https://aur.archlinux.org/packages/8192cu-dkms/

ccrisan, 2016-02-16:
First of all, if it's loaded, it's most likely used. Otherwise it wouldn't be loaded, as there's no command to load it explicitly.

Second of all, assuming you're referring to motionEyeOS, even though you opened the issue for motionEye, the kernel that motionEyeOS uses for the Raspberry PI is 4.1. It's in fact the exact kernel that Raspbian uses, with the exact version of this driver. So the issue can't be fixed in a kernel *way newer* than mine, as the kernel I use is quite new itself.

There's one thing I will never do about motionEyeOS and that is managing with out-of-tree drivers. I always use the latest "official" kernel for each supported board, with whatever drivers it has. This being said, if you really think the driver for 7811Un needs to be updated, please report this to the Raspbian kernel guys and I'll make sure to do a release that includes their updated kernel.

closed 2016-02-16 by ccrisan