motioneye-project/motioneye #126: Thumbnails on recorded videos

okotgnal, 2016-02-17:
i have installed motioneye in a raspberry, everything is working except the thumbnails on recorded videos, i can't see a frame picture of video recorded. Have some way to fix?

ccrisan, 2016-02-17:
Please attach your motioneye.log file.

okotgnal, 2016-02-17:
Hi ccrisan, i don't find this log, is in /var/log? i just find the motion.log, i installed manually the motioneye for raspberry b+ 1.2 
[image]
[image]

okotgnal, 2016-02-17:
Hi ccrisan, i solved it, i install libavresample2, and worked!

closed 2016-02-17 by okotgnal