motioneye-project/motioneye #128: Local webcam not configurable or my config mess-up?

Puzzlr, 2016-02-17:
I have motion ( motion Version 3.2.12+git20140228, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen ) running on Ubuntu 15.10. It can successfully serve a Foscam FI9831P camera and a locally connected Logitec Quickcam Pro 5000 webcam via their respective ports.
However when using Motioneye ( motionEye 0.30 ) I can add de Foscam but am unable to add de webcam (local camera, "no cameras").
Should motioneye automatically use the thread1.conf (etc) as included in motion.conf? I didn't after my setup.
What could be the problem?

ccrisan, 2016-02-18:
What camera devices do you have connected to your system? What does `v4l2-ctl --list-devices` say? motionEye ignores any conf files that come with the motion package.

Puzzlr, 2016-02-18:
UVC Camera (046d:08c5) (usb-0000:00:1d.7-8):
	/dev/video0

Puzzlr, 2016-02-18:
It now simply works:

I renamed all my working motion.conf, thread1.conf and thread2.conf files and installed a standard motion.conf. Restarted motion and motioneye. Voila! Motioneye offered me a UVC camera as input option for a local camera.

I took these steps bases on your answer ("motionEye ignores any conf files that come with the motion package") and realizing that when running zoneminder and motion together I also could not get motion to display the webcam (for this to work the motion website advises to use the video loopback device).

Might I suggest adding to the installation wiki that 'custom made configuration files for motion be removed and replaced by the standard ones for motioneye to work "out of the box" '.

ccrisan, 2016-02-18:
Yes, I should mention that somewhere. I'm glad you managed to get it working.

closed 2016-02-18 by ccrisan