motioneye-project/motioneye #129: Performance improvement of _list_media_files

rolandmoser, 2016-02-17:
The patch below replaces os.walk with os.listdir and os.stat effectively improving the speed of the scanning process by a factor of 2. The patch is against release 0.30 and was tested on a raspberry pi. The overall performance improvement on a raspberry pi with 30000 images was about 30-35%

```
--- mediafiles.py.orig  2015-12-26 15:27:12.000000000 +0100
+++ mediafiles.py 2016-02-17 20:07:59.413857002 +0100
@@ -48,6 +48,17 @@ _prepared_files = {}
 _timelapse_process = None
 _timelapse_data = None

+def findfiles(path):
+    files = []
+    for name in os.listdir(path):
+        pathname = os.path.join(path, name)
+        st = os.stat(pathname)
+        mode = st.st_mode
+        if stat.S_ISDIR(mode):
+            files.extend(findfiles(pathname))
+        elif stat.S_ISREG(mode):
+            files.append((pathname, name, st))
+    return files

 def _list_media_files(dir, exts, prefix=None):
     media_files = []
@@ -79,27 +90,18 @@ def _list_media_files(dir, exts, prefix=
             media_files.append((full_path, st))

     else:
-        for root, dirs, files in os.walk(dir):  # @UnusedVariable # TODO os.walk can be rewritten to return stat info
-            for name in files:
-                if name == 'lastsnap.jpg' or name.startswith('.'): # ignore the lastsnap.jpg and hidden files
-                    continue
-                
-                full_path = os.path.join(root, name)
-                try:
-                    st = os.stat(full_path)
-                
-                except Exception as e:
-                    logging.error('stat failed: ' + unicode(e))
-                    continue
+        for full_path, name, st in findfiles(dir):
+            if name == 'lastsnap.jpg' or name.startswith('.'): # ignore the lastsnap.jpg and hidden files
+                continue

-                if not stat.S_ISREG(st.st_mode): # not a regular file
-                    continue
+            if not stat.S_ISREG(st.st_mode): # not a regular file
+                continue

-                full_path_lower = full_path.lower()
-                if not [e for e in exts if full_path_lower.endswith(e)]:
-                    continue
+            full_path_lower = full_path.lower()
+            if not [e for e in exts if full_path_lower.endswith(e)]:
+                continue

-                media_files.append((full_path, st))
+            media_files.append((full_path, st))

     return media_files

```

ccrisan, 2016-02-18:
Thanks. I've been meaning to do this for some time :) I'll run some tests and do the merge if everything seems fine.

ccrisan, 2016-02-20:
Fixed in c5c66c15a2999b1ede97ae4726c9005858cd1cc9.

closed 2016-02-20 by ccrisan