motioneye-project/motioneye #132: Why no "on_event_stop" action?

Puzzlr, 2016-02-20:
I noticed that under "Motion Notification" you can add a command to be executed. However this is only added to the "on_event_start" directive in de thread-?.conf file.
Luckily, if you manually add a "stop" command to "on_event_stop" directive it does not get overwritten.

I use the start / stop commands to flip a virtual motion detection device in Domoticz (a home automation system) whenever motion decides a motion event either starts or stops, so I can use my IP camera as a motiondetector as well.

Why no seperate options to add a mail / webhook / command option  on event start *and* stop?

ukasiu, 2016-02-20:
I'd also like on_event_stop action.

ccrisan, 2016-02-21:
The reason why on_event_stop is overwritten is that motionEye uses it internally to remove the "motion detected" red frame. Unfortunately there's no way to make it execute custom commands with the current  version. I'll think of a way to allow executing extra commands when the event stops.

maxholgasson, 2016-04-06:
That would be great! I also need it to start a counter and count time since last detection

chrisf4lc0n, 2016-12-16:
Instead of adding it under Motion Notification define it in the Extra Motion Options.
I have done it for the WiFiPlug, so if the feed gets lost it flicks the plug on and off.