motioneye-project/motioneye #133: Capture last motion detected snapshot image.

Status420, 2016-02-21:
I would like to capture the last image that triggered the motion detection and save it to a new location.
If a file could be saved as 'latest' so I could pick it up with a script and SFTP it to my web host.
I'm not doing any port forwarding or connecting from outside, I want to 'push' out the image.

ccrisan, 2016-02-21:
There's a `lastsnap.jpg` file in the root folder of your media files that's a symlink which will always point to the latest taken image. You could use that file to do your SFTP magic.

closed 2016-02-21 by ccrisan

imperialjoy, 2017-04-23:
There is no lastsnap.jpg in /var/lib/motioneye or /var/lib/motioneye/Camera1.  Where would it be by default?

Status420, 2017-08-15:
For my pi, its at /home/pi/motioneye/output

Status420, 2017-08-24:
This ISN'T working, I have still images set to motion and it doesn't create a lastimage.jpg link, can you please reopen and investigate this, it isn't working.
[screenshot motioneye 2017-08-24 17-56-33](https://user-images.githubusercontent.com/10558791/29678826-eb77f92a-88f7-11e7-9be3-05075f7ad421.png)

ccrisan, 2017-08-25:
The lastsnap.jpg file may be removed by the cleanup mechanism. You could use the Video Streaming > Snapshot URL (in conjunction with `curl`, if you're using it locally) obtain the most recent frame when you need it.

Status420, 2017-08-26:
For others also having this issue, i created a script which i run via a cronjob to get the latest image.
`#!/bin/bash
date=$(date +%Y-%m-%d)
input="/home/pi/motioneye/output/$date"
output="/home/pi/motioneye/output"
f=$(ls "$input"/*.jpg -t | head -n1)
ln -s $f "$output"/lastsnap.jpg
`
Hope it helps.