motioneye-project/motioneye #134: Endless reboot loop if output folder was emptied

Gordon-1, 2016-02-23:
(RasPi 2, MotionEyeOS build 20160109)
When all the photos and videos are deleted from /data/output using the "delete all" options in the photo and video media browsers, then the output folder (/data/output) is also deleted. If "/data/output" does not exist when the OS is started the it goes into an endless reboot loop.  Here is part of the motioneye.log file with a comment added about my workaround:
-----------
2016-02-23 17:07:08: [motioneye]     INFO: hello! this is motionEye server 20160109
2016-02-23 17:07:08: [motioneye] CRITICAL: media directory "/data/output" does not exist or is not writable
2016-02-23 17:09:31: [motioneye]     INFO: hello! this is motionEye server 20160109
2016-02-23 17:09:31: [motioneye] CRITICAL: media directory "/data/output" does not exist or is not writable
2016-02-23 17:11:55: [motioneye]     INFO: hello! this is motionEye server 20160109
2016-02-23 17:11:55: [motioneye] CRITICAL: media directory "/data/output" does not exist or is not writable
------
At this point I SSH in and manually created /data/output. Then, in the next restart, the system boot completes...
----------------
2016-02-23 17:14:19: [motioneye]     INFO: hello! this is motionEye server 20160109
2016-02-23 17:14:22: [motioneye]     INFO: cleanup started
2016-02-23 17:14:22: [motioneye]     INFO: wsswitch started
2016-02-23 17:14:22: [motioneye]     INFO: tasks started
2016-02-23 17:14:22: [motioneye]     INFO: mjpg client garbage collector started
2016-02-23 17:14:22: [motioneye]     INFO: smb mounts started
2016-02-23 17:14:22: [motioneye]     INFO: server started
----------------
Note that once the OS is running, deleting all the photos and videos as described, which also deletes /data/output folder, results in this line in the same log:
 [motioneye]    ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/output'

If I understand the code right, this error results from a check to make sure there is free space available for the photo / video save and as long as there is enough space, the output directory is created and the photo and/or video is saved with no problem.

Lastly, note that the media browser title is "Pictures taken (or movies recorded) by cameraX where X is the one chosen for browsing, however is appears to me that it actually lists the pictures (or movies) taken by ALL cameras..

Let me know is you need further.

ccrisan, 2016-02-24:
I'll look into this.

Gordon-1, 2016-02-25:
UPDATE:  This issue actually belongs to motioneyeOS and I now see it has already been reported there but I failed to find it on first search for existing issues.  Since [it was reported here](https://github.com/ccrisan/motioneyeos/issues/86#issuecomment-173760305) I am closing this issue as it is a duplicate report.  The issue is consistent and still exists however.

ccrisan, 2016-02-25:
Let's keep this one opened as the cause might be a bit different. I really need to look into this.

Gordon-1, 2016-02-25:
Sounds good..if you cannot duplicate it let me know and I will provide more info and maybe a video demo.  thanks again, great work!

closed 2016-03-03 by ccrisan

ccrisan, 2016-03-03:
The issue should be gone with the next release.