motioneye-project/motioneye #135: 0.31 stopped sending pictures.

trainspotta, 2016-02-24:
Updated to latest build just now.

I get a motion notification via mail but there are no longer pictures attached.

ccrisan, 2016-02-24:
Alright I'll look into it.

trainspotta, 2016-02-28:
Any news? I'm leaving the house for a few days.

ccrisan, 2016-02-28:
I tried reproducing the issue without success. Please send me your config files (**removing sensitive data**) and I'll try to find the problem.

trainspotta, 2016-03-03:
So now it is getting kind of funny ;)

So I removed the cam completely and set it up again. Now I'm getting mails with pictures again but the working schedule gets completely ignored. No matter if active or not and no matter what time, I always get mails.

What kind of files do you need? motioneye.conf and thread?

Thanks for your work btw.!

ccrisan, 2016-03-03:
thread-1.conf and motion.conf

trainspotta, 2016-03-03:
`# @enabled on
# @show_advanced on
# @normal_password XXXX
# @admin_username XXXX
# @admin_password XXXX
# @normal_username XXXX

thread thread-2.conf
thread thread-1.conf
control_port 7999

control_html_output on
control_localhost on

setup_mode off`

trainspotta, 2016-03-03:
<script src="//pastebin.com/embed_js/n1FBctQn"></script>

trainspotta, 2016-03-03:
<iframe src="//pastebin.com/embed_iframe/n1FBctQn" style="border:none;width:100%"></iframe>

trainspotta, 2016-03-03:
http://pastebin.com/embed/n1FBctQn

ccrisan, 2016-03-03:
How about attaching those files here, directly. I can't seem to find them anywhere in your link (did it expire already?).

trainspotta, 2016-03-04:
So I just deleted every config file, deleted motioneye and reinstalled everything. I jsut finished the configuration. I will attach all files here if this didn't help.

Thank you for your time. motioneye is a really great project!

sherm8n, 2016-03-14:
I am also having this issue.
Was receiving attached images with no problem. Then they suddenly stopped.
I've tried deleting/reattaching the camera, but that didn't help.
In the motioneye log, I get the error:

2016-03-13 19:41:05: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2016-03-13 19:41:06: [sendmail]     INFO: email sent

I still get the email, just with no picture.

ccrisan, 2016-03-14:
The error indicates that there are too many media files to be scanned in due time. Please clean up your media files dir.

trainspotta, 2016-04-07:
Removing and setting up all cameras fixed the issue. Media folder was emptied before.

closed 2016-04-07 by trainspotta