motioneye-project/motioneye #138: Running as normal user

Status420, 2016-02-27:
Hi
I have installed motion eye on my pi using the instructions, now I wished to run the service as a normal user so the files are written as that user and not root (for backup and deletion scripts).
I had everything working as root and all was well.
I followed the instructions to run as normal user but now it will not start.
Nothing in the motioneye logs but in the motion logs I see
httpd_run: motion-httpd failed bind() interface 127.0.0.1 / port 7999, retrying:
[0] [ERR] [STR] httpd_run: motion-httpd socket failed interface 127.0.0.1 / port 7999, retrying: 
[0] [ERR] [STR] httpd_run: motion-httpd failed bind() interface 127.0.0.1 / port 7999, retrying: 
[0] [ERR] [STR] httpd_run: motion-httpd socket failed interface 127.0.0.1 / port 7999, retrying: 
[0] [CRT] [STR] httpd_run: motion-httpd ERROR bind() [interface 127.0.0.1 port 7999]: 
[1] [ERR] [ALL] motion_init: Error capturing first image
[1] [CRT] [STR] http_bindsock: motion-stream bind() failed, retrying: 
[1] [ERR] [STR] http_bindsock: motion-stream socket failed, retrying: 
[1] [CRT] [STR] http_bindsock: motion-stream creating socket/bind ERROR: 
[1] [ERR] [ALL] motion_init: Problem enabling motion-stream server in port 8081:
and so on.
I assume that my 'normal' user needs to added to a group or something, can you point me in the right direction?
Thanks
Leona.

Status420, 2016-02-27:
I think he issue is that I need to give the pi use access to the web server or something, as its not serving the web front end, but so far haven't been able to find this.

ccrisan, 2016-07-03:
Is this still an issue? Sorry for the late reply!

closed 2022-03-19 by MichaIng