motioneye-project/motioneye #139: Multiple Cameras

legotheboss, 2016-02-29:
Hi, 

I upgraded my motioneye software from 0.29 to 0.31.1. My motionEye server connects to two remote motionEye cameras. After the update, I am able to see one camera but unable to see my second camera on the main page... I am able to downgrade and return to the multi-camera view... Both my cameras are remote motionEye cameras. I have also set the settings to show 2 cameras in the layout... What is the solution to this?

ccrisan, 2016-03-01:
Please try removing the cameras completely and adding them from scratch. Config file format has changed a bit and you may be facing some incompatibility issues.

legotheboss, 2016-03-06:
@ccrisan I was able to fix it by updating all my cameras to the latest version. Thanks for your help!

closed 2016-03-06 by legotheboss