motioneye-project/motioneye #140: Unable to add 10th camera to motioneye.

boomerling, 2016-03-03:
I am able to add 9 IP POE megapixel cameras via the interface.  When attempting to add the 10th cameras, all live feeds are knocked offline.  Trying to configure by adding a thread on the backend does not work either.

Is there a restriction to the number of camera's that can be added?
Running a Lenovo TS440 / fedora 23.

Thanks in advance for any guidance you are able to provide.

Note:  I am not familiar with the underlying motion server (have been evaluating motioneye for a couple of weeks).

ccrisan, 2016-03-04:
Please attach your log files here. I personally never tried adding more than 8 cameras to a motion instance. It might be capped at 9? We need to test this. Nevertheless, logs please.

ThomasADavis, 2016-03-10:
It appears to also have problems after 8 cameras, in that you can configure the 9th camera, but no capture occurs.

We want to get to 16-24 cameras in a motioneye instance.  I will be looking into what the problem is - we are still in the process of setting things up.  

I did not see anything obvious in the motion source, or in motioneye (ie, a hard-coded limit/define for number of cameras..)

ccrisan, 2016-03-11:
I'm also interested in this matter myself. However I don't think the browser will be able to render all those 20+ camera frames at once.

fgarpe, 2016-03-22:
I have exactly the same issue, when attempting to add the 10th camera all live feeds are knocked offline. I have just 10 cameras :(, please @ccrisan take a look to this issue, if I can help you only tell me what you need.

arkhan, 2016-06-09:
I have the same problem

dermotduffy, 2016-06-25:
This pull request is relevant to this bug, and allows me to add a 10th camera:

https://github.com/ccrisan/motioneye/pull/233

Thus a workaround to add more cameras would be just to manually set stream_port in the relevant /etc/motioneye/thread-X.conf file to have a sensible port number.

ccrisan, 2016-07-03:
Thanks, @dermotduffy  :)

closed 2016-07-03 by ccrisan